df8oe/UHSDR #1278: Future enhancement if possible. Bluetooth headset.

WM0I, 2017-12-11:
Will there be in the near future the ability to plug in a bluetooth dongle to connect a headset or keyboard?

db4ple, 2017-12-11:
Short: No.
Longer: Someone has to write or port a USB bluetooth stack to do this. Nothing for the faint of hearted people like the current developers and the current software architecture of UHSDR :-( . 
You will have to use the analog line in/out connectors for that purpose.

WM0I, 2017-12-11:
I think I am going to look into using a raspberry pie to interface to the mchf.  I would like to use the rtty function for passing traffic.

Thanks
best regards
WM0I

SergiuEpure, 2017-12-11:
I can connect a keyboard adapter and keyboard itself direct to usb port of mchf and write text

closed 2017-12-13 by WM0I