df8oe/UHSDR #1279: UHSDR Wiki - dead link to KA7OEI SW summary

DF9TS, 2017-12-11:
Wiki ( https://github.com/df8oe/UHSDR/wiki ) contains dead link. See section "Using the UHSDR firmware",  "... you can find here ..." Underlying hyperlink is dead (https://github.com/df8oe/UHSDR/raw/active-devel/mchf-eclipse/useful_manuals/mcHF_Operators_Manual_firmware_ver_0_0_219_26.pdf )

KA7EOI doc can be fand in mchf-eclipse->useful manuals->old manuals.

Minor detail, but consider with low prio. Any newcomer in the project will stumble on such things...

73

Gerd.

db4ple, 2017-12-11:
Hi Gerd, 
you are invited to fix that: the problem is the rename of the repository I think: mchf-eclipse is now UHSDR. And we moved the document to a sub folder in addition (oldguides).

Thank you,
Danilo

DF9TS, 2017-12-11:
Done.

closed 2017-12-11 by DF9TS