df8oe/UHSDR #1280: Bug in AM

ly2bok, 2017-12-11:
Starting with ver.2.5.130 in mcHF AM does not work for automatic gain control. The high signal level rises to the maximum and does not return to its original position. Only after turning the encoder ENC2 returns to normal gain.After a strong AM signal, mcHF loses sensitivity after disconnecting the antenna and connecting it again. And just after turning ENC2 into the encoder, the sensitivity suddenly appears.  In other modes work  is fine. It only concerns AM. Check it out.
Justas LY2BOK

DG9BFC, 2017-12-11:
use newer version to test ... maybe it is an old bug??

ly2bok, 2017-12-11:
No, this error is also in the latest version. We tested in three mcHFs

DD4WH, 2017-12-11:
Hi Justas,
thanks for the report!
Can you give more details on how to reproduce this? I cannot reproduce at the moment on my machine (OVI40 & mcHF RF 0.4)
73 Frank

df8oe, 2017-12-11:
I think it is due to a not ideal setting of AGC. I can reproduce - if I misalign my AGC settings...

DD4WH, 2017-12-11:
hmm, can you tell how I do that: misalign my settings? Do you have a too low setting of AGC thresh?

DD4WH, 2017-12-11:
Ah, OK, just had that effect! Seems I have it occasionally, but can you give a setting that would reliably reproduce it?

DD4WH, 2017-12-11:
hmm, impossible to reproduce again without setting AGC thresh too low . . 
do you have a hint how I can reliably reproduce the problem?

DD4WH, 2017-12-11:
if yes, try to switch it OFF and try if you can reproduce the problem again with fade leveler OFF, please!

ly2bok, 2017-12-11:
"fade leveler" activated in my Standard menu. I try to switch it OFF and try if again.  The problem
do not go away
AGC in window is 10. Someone does not work that way.

DD4WH, 2017-12-11:
is it still there if you increase AGC thresh to 30?

ly2bok, 2017-12-11:
If I make 30 AGC, it's even worse. At the high AM signal, I disconnect the antenna and connect it again to a low volume, but it is enough to rotate the ENC2, the sound suddenly dances upwards.

DD4WH, 2017-12-11:
hmm I do not understand what you mean and cannot reproduce.
Can you explain it systematically, please?

DD4WH, 2017-12-11:
1. you switch on your TRX
2. you tune into a strong AM station
3. what are you doing then?
4. what does "dance upwards" mean?
5. why do you disconnect the antenna?

ly2bok, 2017-12-11:
Sorry for the bad English. This is the translation of GOOGLE.
I turn off the antenna so that there is no signal. When we connect the antenna again, the sound level is small, but when the ENC2 twists, the sound very strong. AGC does not work. It only concerns AM.
Good night

DG9BFC, 2017-12-11:
hmm i can reproduce a "similar effect" ....just tune to a free spot in am .. .then each setting in the agc shoots up volume and then it dros down ... regardless if you change the time ... or strength or "anything in the agc part" the volume jumps up and then goes down if there is a pulse noise or similar noise (you will only find a noise free spot with no antenna ... grin) ... yes there is some effect ... but that is normal ... with no signal there is no agc signal so mchf cranks up gain

ly2bok, 2017-12-12:
There is an error there, but you did not understand what I wanted to say to you. Maybe someone else will find and better tell.
  I close the topic. Thank you and goodbye

closed 2017-12-12 by ly2bok

DD4WH, 2017-12-12:
hi Justas,
thats not very helpful . . .
If you want us to find the bug, please: you need to explain to us in detail, what you really observed!
And you also did not yet answer some of my questions:
1. Which hardware do you use?
2. Did you really exactly find out, when this bug started: was it really with version 2.5.130 ?
3. With your antenna connected and your hardware tuned to only noise, which AGC thresh do you have to adjust too, so that the AGC box switches on?
Thanks in advance!
73 Frank

ly2bok, 2017-12-13:
I have SDRmcHF. 4 version. As I understand you do not have SDRmcHFF. I have a processor STM32F407, the memory is 24LC1025. software ver. 2.7.42. Problem AGC in AM mode from ver. 2.5.131. in ver. 2.5.130 everything is good
    With my antenna connected and I hardware tuned to only noise in AM, AGC box switches on 22-23
 Example:
1.Tune receiver to the strong AM station.
2. Walk  from the station about 10-15 Khz
3. Rotate one step ENC2.
4. Listen to how the receiver's sensitivity will rise.
Of course, that the AGC has not returned to the desired level and the receiver has decreased its sensitivity until it turns in any direction of the ENC2..
This is very fine see  when strong short spike in frequency. The receiver sensitivity decreases sharply.
LY2BOK Justas

DD4WH, 2017-12-13:
Hi Justas,
thanks a lot for the detailed report! That´s very helpful!
I will dig deeper in the code and will report back as soon as I find something!
All the best 73s Frank

DD4WH, 2017-12-13:
Hi Justas,
one last question:
when you use the old firmware version 2.5.130 (which works well for you as you said), do you have WDSP AGC enabled in the standard menu or do you use the standard AGC?
That could make the difference in reaction of the AGC!
The old (standard) AGC is available up to version 2.5.130 and NOT available any more in all version from 2.5.131 upwards.
73 Frank

ly2bok, 2017-12-13:
The 2.5.130 version has a STANDART AGC and WDSP AGC. Both options work fine.
1. STANDART AGC - works
2. WDSP AGC - works
Preferably work with STANDARD AGC
73! LY2BOK Justas

df8oe, 2017-12-13:
The issue can probably be a setting regarding "old NR" or "old AGC". Justas has another issue because of using the old AGC some days before (and maybe changed settings for old NR, too)... I haven't used old NR from the beginning because of it is *absolute useless" and switched to new WDSP AGC since it is implemented (because it works muuuuuch better than old AGC)...

You can test if this causes issues by loading default settings at boot and check again if issue is present. Old setings are not overwritten until you do that explicitly or switch off mcHF.

Sorry - old AGC is OBSOLETE. There is nothing new WDSP AGC cannot do better...

DD4WH, 2017-12-13:
I did check the code changes from version 2.5.130 to 2.5.131 and could not find anything that could cause something wrong . . .
Maybe you could try to go step-by-step through the adjustment procedure of the WDSP AGC with the newest firmware version. Then you could test whether the issue stays the same after having adjusted the AGC:
https://github.com/df8oe/UHSDR/wiki/Automatic-Gain-Control-(AGC)
73 Frank

DD4WH, 2017-12-13:
Thank you Justas and all others for your patience!
I found the error in the AGC code and will post a pull request in the next days!
73 Frank

DD4WH, 2017-12-14:
FIXED

ly2bok, 2017-12-14:
Good day, Frank.
Ok, I already checked version 5.7.44. The AGC in AM fashion works great. What and was required to see .Thank you for the work done.
We will look forward to when you make a good  DSP A-NOCH
73 Justas

DD4WH, 2017-12-14:
Yes, we are busy on autonotch :-).
And I look forward to receiving many detailed error reports from your side ;-). 
We are really dependent on people like you who really test intensively and give good descriptions of the issues, so we all can contribute to optimize the UHSDR software.
Thanks and have fun,
73 Frank