df8oe/UHSDR #1281: Noise floor dramatically increases with digital out but if DIG I/Q?

Tibet4690, 2017-12-12:
If this is a noob question feel free to tell me so, I'm still learning.

When using fldigi or wsjt-x my noise floor increases dramatically. 

This seems to be tied to using the mic/line select "digital" setting.

I at first thought this was due to RFI from my laptop, but now my novice opinion has changed... I fooled around with HDSDR with omrig. When using this setup I did not experience the same change in my noise floor even though I am using the same laptop, usb cable, radio, etc. 

Can anyone explain this to me? 

Thanks for the input

df8oe, 2017-12-12:
Sorry - I do not undrstand.

If you use SDR software like HDSDR or QUISK you need to set DIQ (which means "digital out i/q") and if you want to use digital modes software like fldigi, MSSTV and so on you must use DIG (which means "analog audio via USB"). Of course you cannot compare the noise floor of both because one is audio and the other one is I/Q.

73
Andreas

DG9BFC, 2017-12-13:
9hmm just a guess ... in fldigi the cat is used for ptt and frequency??? maybe the cat commands are producing traffic on usb cable (polling of freq each half second or so) ... do you also use cat in hdsdr (sync frequency of mchf and hdsdr)??? there you have the omnirig settings (where you can select poll intervals etc.)
if you do not use cat in hdsdr ... then we found the noise source (usb traffic)
try with some ferrites ... a longer cable and some distance plus ferrites may help a bit
... i can only tell that from my tests with different sdr on usb port of my laptop
best results (lowest noise) was with a bit longer cable, some ferrite cores ... and the sdr grounded on the antenna input (and laptop floating rf wise cause it has no direct ground connectin elsewhere)

Tibet4690, 2017-12-13:
After experimenting some more last night I think you are right, this has to do with rig control, the interference appears to be there on hdsdr when using omnirig

I had some clip on ferrites on the cable but the noise still was there. I'm going to try looping the cable through some larger clip on ferrites to see if that has any effect.

You mentioned the polling rate can be changed... should I be playing with that to see if I can mitigate the interference that way?

I am in an apartment so grounding can be a little interesting... outside of that, I'm operating portable.

DG9BFC, 2017-12-14:
so at least we found the noise source :)
to get rid of that noise ... is then on your end
if ferrites plus longer cable does nothing ... then your usb noise gets into the rig via the antenna (you said you live in an apartment so i am guessing your antenna is close to shack??)
good luck
ps an you test with another pc?? other laptop borrowed from a friend??

Tibet4690, 2017-12-16:
Great info thanks, lots to learn.

Yeah, I've tested with my desktop and laptop. So doesn't seem to be machine specific. 

I've been playing around with a random antenna to an MFJ QRP tuner, but I've also played around with other antennas with the same result.

DG9BFC, 2017-12-16:
also checked with a dummy??? does it get into mchf via the usb cable (common mode current/noise) or via antenna ??? ... you said you tested that vith a random wire plus tuner ... that is the worst setup for noise cause antenna is as close as it can get with such a setup
is it the same on all bands?? same on upper and lower end of a band with high noise ??
if you have that noise ... with a dummy load and ferrites on the cable ... then i am guessing you have to live with it ...

DG9BFC, 2018-01-07:
can we close that issue?!?

closed 2018-01-21 by df8oe