df8oe/UHSDR #1282: Suggestion: Custom Sidebar for Wiki, structured Document view

DF9TS, 2017-12-12:
Please have a look here for a potential custom sidebar for the Wiki, in order to get documents structured a bit:
https://github.com/DF9TS/UHSDR-1/wiki

Links point to df8oe GH, so could be simply copied in.

db4ple, 2017-12-12:
Hi Gerd,
that is awesome. Did not knew, Wiki can do this. Do we write this once and can use it on many pages?

df8oe, 2017-12-12:
Hi Gerd,

many thanks for the hint! Much better than original. I have implemented it already.

DF9TS, 2017-12-12:
Thanks Andreas. Helps me read through the wiki.

So yes Danilo: this is static HTML lists and links, and for a new wiki page added the HTML list needs to be extended in the custom sidebar, and only the admin can do it.

df8oe, 2017-12-12:
Hi Gerd,

that was completely new for me. Do you know if it is possible that "contributors" can edit the sidebar, too?

db4ple, 2017-12-12:
Yes, I can edit it. Did not do it, though.

DF9TS, 2017-12-12:
Also can edit it.  Can see "pencil" icon in custom sidebar, next to "home".

On second thought: Could be better to only have the main sections in sidebar ("Supported HW", "Building mcHF" etc) as list without subtopics. Clicking on main topic will open wiki page per main topic. Sub topics can be placed in these wiki pages.

Advantage: Easier management of new or changed sub topics directly with wiki editor. More place for sub topics in the separate sub pages (titles can be much longer) and still a very clear and concise main custom sidebar.

DF9TS, 2017-12-12:
Have a look here: https://github.com/DF9TS/UHSDR-1/wiki

Custom side bar on right for top level wiki subjects.

All sub topics  on sub pages. 

Could be better, given that wiki will grow?

DF9TS, 2017-12-13:
Done

closed 2017-12-13 by DF9TS

DD4WH, 2017-12-14:
Hi Gerd,

great idea to structure the WIKI!

But please show all the sub-pages in the sidebar! 

Otherwise I am not able to find even the WIKI pages I set up . . . We do not have so many pages yet, that this could be called useful.
AND: so few people use the Wiki pages, that hiding the subpages will surely contribute further to people not finding what they are looking for.
So it would be very nice, if you could revert only the last change you did: show all the subpages in the sidebar.
Thank you very much!
73 Frank

df8oe, 2017-12-14:
Hi Frank,

new sidebar is very well structured and you can get pages related to every main title by one click. OK - if you do not know where the document was placed you have to click on every title. But there is an easy solution! Just click on the small arrow left of "Pages" above new sidebar and you do have "old structure!" So I do not see any advantage to revert anything. The ones who like small and effcient listing can use new sidebar and the ones who like to see all existing pages can use "Pages" feature :)

73
Andreas

DD4WH, 2017-12-14:
Hi Andreas,

at the moment I can only see the main titles. All the pages are hidden. That is useless, because you have to click on every main title to see whats behind it. And even worse: if you click on it, the contents do not appear under the main title, but on the main page at the left.

And I have no small arrow to click on . . .

At the moment the Wiki is not usable any more for more, if it stays like this.

73 Frank

df8oe, 2017-12-14:
> At the moment the Wiki is not usable any more for me, if it stays like this.

That is very sadly, because of 99% of other users will see that 180° other direction.... So I would appreciate much that new structure stays as it is and have a look why you cannot open "all pages".
[image]

I attach a screenshot where you can open "old style sidebar".

73
Andreas

DD4WH, 2017-12-14:
Hi Andreas,
thanks a lot for the hint, did not see that button.

However, its still not usable any more for me: I see a list of the pages, which is unstructured (that is the same as we had for months). That is exactly what Gerd wanted to improve and the structured list in the sidebar is really awesome.

The situation at the moment is:
* I see a list of topics (no pages are visible)
* I do not know which page is behind which topic
* I have to click several times to look at the pages behind the topics
* the pages do not appear under the topic, but on the left side on the main page

This is not an improvement.

Please revert to a structured list of pages in the sidebar, that is the optimal state, so you can immediately see the topics with the corresponding pages that belong to each topic.

All the best 73 Frank

DF9TS, 2017-12-14:
No worries.

Will revert to old sidebar (Topics and all subtopics) as was before.

Will put the new "short" sidebar (just the main topics) on top of that previous, so we can see how that looks (Number of wiki pages should increase). 

Will do this the next 2..3 days as I am currently "busy" so please be patient, tnx.

One question to the gurus: The wiki is a "UHSDR" wiki, or a "OVI40" wiki also? Should it strictly handle SW or will it target HW also (some mcHF articles there now, what about upcoming OVI40...).
Wiki structure would depend on direction taken...

Gruß

Gerd.

df8oe, 2017-12-14:
Hi Gerd,

this is intended only to be UHSDR WIKI. All OVI40 related things should be placed to amateurfunk-sulingen WIKI.

BTW:
Do you not see the button to expand all pages?

73
Andreas

df8oe, 2017-12-14:
And additionally:
New structure on top is appreciated :)

DF9TS, 2017-12-14:
Have a look. OK now?

3 ways to access Wiki pages:
1) expand button (yes I see it)
2) sidebar short menu, then wiki pages via topic page
3) sidebar with topics and subtopics

df8oe, 2017-12-14:
Hi Gerd,

that's why I am wondering... Why do you see that button and Frank does not? It is not "hideable"!

For me it's ok now. The very pragmatic "short form" on top - the rest downwards.