df8oe/UHSDR #1283: What is the best way to connect a desk microphone and a foot switch?  QUESTION

WM0I, 2017-12-12:
I am  watercolorist and I want to do my art along with talking on HF. I am going to connect the footswitch to the straight key input but I need to know about what types of microphones I can use. 

Best regards
WM0I

db4ple, 2017-12-12:
Electret condenser microphones (dynamic microphones don't like the bias voltage). Majority of PC headset microphones and (cheap) radio microphones are of this kind. Basically any microphone you can connect to a PC's analog microphone input should work without problems.

df8oe, 2017-12-12:
I recommend dynamic microphones. I have tried many different mics but the best results are with dynamic ones... Not mandatory - it's my investigation.

73
Andreas

DG9BFC, 2017-12-13:
computer headset ... that way you have headphone ... and mic is always at same distance

mariomax232, 2017-12-21:
Hello...I conect by Line IN  dynamic microphone+preamp+parametric equalizer+sound realzer  and the TX audio is broadcasting!!
73...EA7IXD

mariomax232, 2017-12-21:
If it had the option to modificate the audio bandwidth in TX the sound is better.

mariomax232, 2017-12-21:
If mcHF had the option in the menú of equalizer the audio TX ...the sound it would be better more...

WM0I, 2017-12-24:
I know if I use a dynamic microphone with 600 ohm impedance and is connected to the MIC jack I have to use  capacitor of about .1uf or a 1uf small ellectrictic to isolate the dc voltage from the Mic. Question is if I connect the dynamic Mic to the line in do I need the additional CSP?

Also do I need an equalizer? 

Best regards
WM0I

mariomax232, 2017-12-24:
I think if you conect the dynamic mic by a LINE IN Jack  is more easy...no need capacitor to block the dc voltaje.
Is my opinión .....I have my mchf conected mic dynamic  by line in Jack and sound very good.....but I have preamp and parametric equalizer ....I like the HIFI SSB.
EA7IXD....73!!!

DG9BFC, 2017-12-24:
mic in has bias resistor (desolder it and you need no coupling cap)
line in has no bias resistor and no bias voltage ... so no coupling cap needed
actually there ARE already coupling caps in all inputs so adding another one in line would just give more high tones (or lower basses) ...
you may experiment with an additional cap to somewhat lift up high tones (if you have a low tone voice and/or mic) ... personally i also like to hear those "bassy" ssb sound that some stations send out BUT low tones only takes much transmit power (needs much energy) but does not give much to readability on dx ... and the mchf is a qrp rig so the low tones (tx filter closer to carrier spot ... and low tones lift up with an equalizer) does only takes energy without giving you more talking distance ....
i would NOT lift up the basses tooo much .... i already have a bassy voice and i would have to lift up the higher and mid voice tones to have better readability
we have the tx filter selectable in the menue ... if that is not ok for you then you may experiment with additional audio filtering in front of the mchf (but as already noted ... basses only costs energy!)

EA3BDD, 2018-02-09:
Hi all.
I use from ver 4.0 of mcHF SDR, the Astatic D-104 with T-TUG8 without problems !!.
I use this microphone ( 5 units ) with all my rigs.
No modifications in the circuits.
Josep