df8oe/UHSDR #1287: 2.7.43 noise reduction= good. Now if we can get notch filter back

WM0I, 2017-12-13:
I am now starting to hear small signals I wasn't able to hear before because of the noise. A notch filter would be a great addition to to the new versions.

DG9BFC, 2017-12-13:
yes please ... old notchfilter back and old nr also selectable :)
PLUS adding the new nr with the new vad
or in other words back to version 2.7.31 .. but with the new zcr vad added to menue

df8oe, 2017-12-13:
The new VAD cannot work with old ugly NR - it's a complete other technic ;)

Old notchfilter also cannot work togehter with very good new NR - please stay tuned. There will be another approach which need not go back to "the old cruncher"...

DD4WH, 2017-12-13:
Thanks for the feedback!
Our goal is 
1. to provide a very good NR
2. to provide a very good autonotch filter

->1. I think we have reached a quite nice result, but we are still working hard (which you will not notice, because we do it without git ;-)) at optimizing the result. The amount of scientific/engineering literature in this noise reduction field is so huge, that it will take us some more weeks/ probably months to obtain a satisfying result. So please be patient, I am sure we will be able to deliver a nice result.

-> 2. I am aware that my first results in this field are very unsatisfying. I have a totally new approach ready, but it does not work yet :-). Please give me another two or three weeks for a first test version. If we will not manage to get the autonotch running at least as good as the old autonotch, I will re-install the old autonotch, promised (but only after we tested at least another two months) :-).

73 Frank

DG9BFC, 2017-12-14:
andreas ... i know that you hate the old cruncher :) ... but i also know that new nr works superb together with the old notch filter ... and in some cases and with finetuning of the menue settings you can have old notch with new nr ... or new nr plus old nr (yes in your ears ugly sound) ... i have tested old and new nr intensively (with all possible combinations) ... and with some noisy signals i used BOTH nr together ...
old nr alone does not work nearly as good as the new one ... that is correct ... and new notchfilter has to be developed
what i was asking was ... old notchfilter (or new one with same performance) ... with new nr and the (now new) zcr vad (and if the old notch can not be used without having also the old nr in the fw then keep old nr)
if we have a better (or similar good) working notch ... only then we should delete the old notch filter completely (and frank promised to reinstall the old notch if he can not make a better one or in minimum a similar good one)
frank is right ... we have a very good working nr now (a tiny bit better with zcr vad) ... i will try all versions like i did before (and report my results as usual) ... and hopefully he can develop a functional notch in a few weeks (would make a nice xmas gift ... a good working notch filter ... grin)
if that takes one or two months ... ok so be it (remember what we had in the fw a year or two back???)
73 sigi

DD4WH, 2017-12-14:
Maybe it is good to know that the ONLY reason to deinstall the OLD autonotch filter (before the new one was ready . . . ) was to free some RAM so that the NEW noise reduction AND the NEW spectrum & waterfall display were allowed to work on the small RAM (192k) mcHF machines.
Hopefully this knowledge makes it easier for you to wait for the developments ;-).

DG9BFC, 2017-12-14:
it is easy to wait even without knowing the details :)

S52CQ, 2017-12-18:
In the mean time manual Notch does work well.

closed 2018-01-12 by WM0I