df8oe/UHSDR #1292: Version 2.7.45   "fonts"   Some like / some don't .

WM0I, 2017-12-14:
Version 2.7.45   "fonts"   Some like / some don't. All in the eye of the beholder I guess.I personally like the old font. But after reading a few of the comments from McHF groups the new font is what they like. I know there is not enough room in the ram to do both. Question can two releases be made for each font?  2.7.45a and 2.7.45b. Just a thought.

Best regards
WM0I

sp9bsl, 2017-12-14:
I vote to revert to old fonts. We should save the flash space for other things that can be really helpful. It doesn't look good on 480x320 screen (hard readable).

73 Slawek

WM0I, 2017-12-14:
I agree, New font is hard to read. I vote to keep the older font.

KJ6ZH, 2017-12-15:
Do not like that new font.  Terrible on the eyes.
Previous version is clean, easy and quick to read.

df8oe, 2017-12-15:
I was much exited about the reactions (of course it is absolutely simple to switch back). There will be much more features in future which "some like and some don't". I am thinking about how to handle this. Actually we need a decision and proceed with one of the alternatatives. But that must not be forever this way ;)

db4ple, 2017-12-15:
I vote to disable the new font (not to remove it). With fixed-size fonts we can make the font set selectable relatively easily (as long as the new font set uses the same character sizes). 

If someone wants to do this as a learning excercise for UHSDR development/GitHub, I am happy to help...
73
Danilo

df8oe, 2017-12-15:
I have already done that.

db4ple, 2017-12-15:
What? Dynamic selection of font set via menu or switching of the new font?

df8oe, 2017-12-15:
Disabling the new font.

db4ple, 2017-12-15:
Ok, of course 👍 My offer was for the second part (dynamic selection of font sets via menu).

sp9bsl, 2017-12-15:
I think that woludn't be good to have a folder (for example in official release zip file) with pre alfa when we want to publish something contentious (like the visual aspect) or have some kind of sanbox, just to make some kind of survey to know what most users prefer

maaberu, 2017-12-15:
Thank you all.

DF9TS, 2017-12-16:
@Danilo: Can volunteer to do this (dynamic selection of font sets via menu) in case this can wait a bit.
Waiting for the pending hardware delivery (OVI40 UI kit &  dispay). So could start in January.

db4ple, 2017-12-16:
Let me know, when you're ready.
73
Danilo

DG9BFC, 2017-12-16:
nice would be if we had an online firmware builder and a homepage where a user can select what gimmicks he wants in the fw and what not and maybe also what cpu he has ... select the functions ... press on "build now" ... and then he gets a binfile built for his needs (and if he has selected too many thingys with a warning that it will only work in bigger cpu') ....

closed 2017-12-24 by WM0I