df8oe/UHSDR #1299: This will be the biggest question I have to date. Special Firmware development

WM0I, 2017-12-20:
I have no Idea what it would take to create a special version of the firmware to go through as many circuits as possible to show problem areas. This would help in correcting and or fixing a problem such as the one I am having now. Problem with adjusting PA output for each band after setting the PA bais. I have set each band 5 Watt setting to the value of 3 and I still get full power of over 15 watts.  Plus I am getting TX audio distortion on all bands with the MIC setting of 10 to 17.      

Can there be a firmware designed to help shoot?

Best regards
WM0I

df8oe, 2017-12-20:
Please follow _exactly_ the steps that are described in WIKI. If you get full power on 5W with 5W PWR setting to "3" I am absolutely sure **you do not tune / transmit with 5W**. You tune with FULL PWR, due to your setting "tune power level" :)

It is impossible for firmware to "go through circuits and show problems" except the ones which are already implemented. Firmware already identifies hardware communication errors on I2C or SPI bus and reports them. More is impossible.

TX audio distortions can have many cuases. Mostly PIN diode switching is the issue - but that is only a guess.

73
Andreas

WM0I, 2017-12-20:
There are no PIN Diodes to do the switching. The board I have have a revised PCB that has the relay circuits on the board and mounted with the transistor and shunt diode already. The distortion sounds wobbly like being under water.

df8oe, 2017-12-20:
And they are dependent to mic gain setting? Only in a defined span - below and above they're gone?

WM0I, 2017-12-20:
No, Below 10 the audio is almost gone altogether. Above 17 the audio is so load it sounds like it nothing but distortion with no understanding what is being said. My unit is turning into a 10 to 15 watt CW machine. LOL

df8oe, 2017-12-20:
That's curious at all... This is 100% a hardware issue. It can be a wrong wired mic, an incompatible mic or a soldering issue in the neigbourhood of audio codec / audio switching.

hazzer23, 2017-12-21:
Hello, I've found I had to select "low power" in the PA config menu for both low and hi band before I could set the RF to 5w, the clone radio PA stage seems to be easily over driven, once 5w was set on all bands then the full power could be set to no more than 10w (8w on 10m). I'd suggest trying these settings, as I'd expect 15w to be quire distorted.

WM0I, 2017-12-22:
I found a different method in adjusting the PA for each band. Set the PA Bias to some value above 85. then set the values for each band to 3. Set the tune level to 5ws and select the power output level to 5ws using the m3 (PA) on the front panel. now go to the PA config section for each band and adjust each band by clicking the PTT switch on the hand mic. Speak into the mic with a normal volume that you would in a normal SSB conversation and adjust each band until a power meter swings up to the 5 watts on a external meter reading. Stop speaking to make sure the needle on the power meter falls back to zero watts.  

Do this for each power level for each band.
You can do the same method  for the full power levels across all bands.

This has seemed to worked for me so far. Now to make contact on SSB for audio TX quality. 
I will report my finds. 

Best regards
WM0I

closed 2017-12-22 by WM0I