df8oe/UHSDR #1300: I think I found a better way.

WM0I, 2017-12-22:
I found a different method in adjusting the PA for each band. Set the PA Bias to some value above 85. then set the values for each band to 3. Set the tune level to 5ws and select the power output level to 5ws using the m3 (PA) on the front panel. now go to the PA config section for each band and adjust each band by clicking the PTT switch on the hand mic. Speak into the mic with a normal volume that you would in a normal SSB conversation and adjust each band until a power meter swings up to the 5 watts on a external meter reading. Stop speaking to make sure the needle on the power meter falls back to zero watts.

Do this for each power level for each band.
You can do the same method for the full power levels across all bands.

This has seemed to worked for me so far. Now to make contact on SSB for audio TX quality.
I will report my finds.

Best regards
WM0I

df8oe, 2017-12-22:
This way is not a solution at all - it is an approach...

If you are adjusting to speech level dependent output your adjustment shoes an average of power which depends on many different factors (frequency of your voice, volume of your speech, mic level, compressor level e.g.). Your peak output is much higher than the "5W" or "10W" @ FULL. If you adjust power output this way you will overdrive your PA which results in distortions. I strongly recommend to use the way described in GitHub WIKI and take "TUNE" button for generating a well defined tone for that calibration.

73
Andreas

db4ple, 2017-12-22:
Hi WM0I,

thanks for helping us to improve the firmware and associated documentation for procedures. As Andreas, DF8OE said, in this particular case there is no alternative to the procedure described in the Wiki. If the instructions are unclear, let us work this out and improve the wording. The procedure itself is proven to lead to repeatable results and is also technically sound. 

However, describing such a procedure in a way that it is easy to understand and follow is a different matter. And here we could use your help. Please read the instructions carefully and let us know where it is unclear/ambigous and we can improve.

Another suggestions: The GitHub issues should have a topic identifying as best as possible what the issue is about. So for this one you could have used something like "Better way to calibrate PA power", so everyone would know what the issue is about and can ignore it if not interested.
 
One final comment, please understand that GitHub issues are not intended to be a replacement for the forum / newsgroup use. If it is clearly an problem identified in the firmware, its related documentation, or a requested feature or something like that, the best place is to go straight here. 

If it is a general question or discussion point to be clarified for one's own personal knowledge/use, please use the available discussion forum/news groups first before deciding to go here and creating an issue. 

Thank you and 73
Danilo

closed 2017-12-22 by WM0I