df8oe/UHSDR #1305: New touch calibration: drawing test on 3,2" TFT does not work correctly

DL8EBD, 2017-12-25:
with 3,2" TFT the touch calibration works ufb but the drawing test at the end does not work correctly.
It only works on the upper half of the display and the drawing line is havy speckled (dotted)

There is no problem with 2,8" or 3,5" displays

[image]

sp9bsl, 2017-12-26:
Hi Thomas,
I found something, let's see if it helps.  #1307 

73 Slawek

DL8EBD, 2017-12-26:
Hi Sławek,
yessss, it helps!
Drawing test now works over the complete display area.
It's still speckled, but this is only cosmetic.
Touch is very precise at all points of the TFT
many thanks
Thomas

[image]

df8oe, 2018-01-21:
Please close this issue.

closed 2018-01-21 by DL8EBD