df8oe/UHSDR #1306: 2.7.49? Question-----What is the purpose of the update?

WM0I, 2017-12-25:
I read all the comments for each line of code and all I could get out of it was new documentation for the menu structure. Did I miss something? I just went back the code and saw nothing other than menu structure.

df8oe, 2017-12-25:
We are discussing UHSDR related themes in our discussion forum - section "UHSDR". You can read there - and if you want to write something feel free to register...

https://www.amateurfunk-sulingen.de/forum/index.php?board=19;action=display;threadid=905

DG9BFC, 2017-12-26:
or in short ... version 2.7.49 adds touch screen calibration

sp9bsl, 2017-12-26:
I explained the purpose of the changes here: #1310 .
Please close this issue.

closed 2017-12-26 by WM0I