df8oe/UHSDR #1308: CAT enhancement

quadrant2005, 2017-12-26:
hi there here are some hidden CAT commands i thought you may find useful in the future...

  ?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbb } }, /* eeprom read */
    //?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbc } }, /*[QBS] eeprom write (Undocumented) */
    //?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xa7 } }, /*[QBS] read radio configuration (Undocumented) */
    //?{ 0, { 0x00, 0x00, 0x00, 0x00, 0x10 } }, /*[QBS] Read TX keyed state (Undocumented) */
    //?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbd } }, /*[QBS] Read TX metering  (Undocumented) */ 
    //?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbe } }, /*[QBS] Full Factory Reset (Undocumented) */

df8oe, 2017-12-28:
No. As you can see they are _undocumented_ and so not reliable in use.

db4ple, 2017-12-28:
Thanks,
don't know I know why you closed #1309 right after opening it, but anyway: we already support
0xbb/0xbc/0xbd and 0xa7 (https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/cat/cat_driver.c#L704) which is used by some
For 0x10 and 0xbe we have no support but I am also not sure if there is any program out there using these.

db4ple, 2017-12-28:
@df8oe: The undocumented commands are there (unchanged) since the original FT817 many years ago so I think we can safely assume these are there to stay. And the ones we implemented are used in almost all CAT programs including hamlib.

quadrant2005, 2017-12-29:
hi guys, i have documentation on how to use these commands if that would be helpful. Are you guys looking to make the Mchf yaesu ft817 CAT compatible etc or FT857/FT897 those two radios use a different memory map. Has there been any thoughts on using a fresh CAT control command set instead then request Chirp to add support for it. At home ive opted to to use the Elecraft K3 Cat control command set as its much simpler. Would be nice to hear feedback from you guys. It would be very nice to control every single feature that can be changed from the menu also via CAT control.  Seasons greetings to you guys all the best for the coming new year.   HamRadio Deluxe makes use of the undocumented commands, The Yaesu FT857 faceplate command set has been reverse engineered to guys, so every single thing that can be set from the faceplate has now been worked out the changes are written to the eeprom. I did not realise that i have not worked with the clone command of the FT857 radios but i do have several sets here and a Rigol scope that has serial decoding so after the new year i could look into this if you guys are interested. My Mchf dev unit has a STM32F427VIT6 with 2mb Nandflash and im using two eeproms setup as 4mb addressable   FT817 commander is very useful and was written by the creater of hamradio deluxe before it was sold on. Its been useful for when i was reverse engineering to get a arduino to emulate a ft875 radio cat..

df8oe, 2017-12-29:
Not a problem. We are seting labels. And using GitHub is not quite simple - but not to complicate :)

I do not miss any command in CAT - please be precise. What new commands do you want to get?

DG9BFC, 2018-01-04:
maybe the "read radio configuration command" could be used to save configuration (menue settings) in the pc???

db4ple, 2018-01-04:
No,  this command is not what you think it is and it is already implemented and used (since some CAT programs make use of it).

73
Danilo

DG9BFC, 2018-01-04:
ok ... just a thought