df8oe/UHSDR #1309: Cat Control

quadrant2005, 2017-12-26:
hi there here are some hidden CAT commands i thought you may find useful in the future...

?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbb } }, /* eeprom read /
//?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbc } }, /[QBS] eeprom write (Undocumented) /
//?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xa7 } }, /[QBS] read radio configuration (Undocumented) /
//?{ 0, { 0x00, 0x00, 0x00, 0x00, 0x10 } }, /[QBS] Read TX keyed state (Undocumented) /
//?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbd } }, /[QBS] Read TX metering (Undocumented) /
//?{ 0, { 0x00, 0x00, 0x00, 0x00, 0xbe } }, /[QBS] Full Factory Reset (Undocumented) */

closed 2017-12-26 by quadrant2005

db4ple, 2017-12-26:
Thanks,
don't know why you closed it right after opening it, but anyway: we already support 
0xbb/0xbc/0xbd and 0xa7 (https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/drivers/cat/cat_driver.c#L704) which is used by some 
For 0x10 and 0xbe  we have no support but I am also not sure if there is any program out there using these.

quadrant2005, 2017-12-29:
hi there db4ple, i accidently duplicated my post while trying to work out how to set a label, I still do not know how to attach a label :) unless it is you guys who do that from your end. I was not sure people could see my post without a label. Im very new to using github. Thank you for your patients