df8oe/UHSDR #1310: New touchscreen functionality

sp9bsl, 2017-12-26:
Hi, 
starting from the 2.7.49 we implemented completelly different approach to touchscreen. It is now mandatory to use calibration at powerup before use, the old calibration will not work.
So please:

**TO ENTER CALIBRATION, PRESS AND HOLD THE TOUCHSCREEN AT POWERUP** 

and follow the steps shown on the  screen. After the calibration finished and the rig works normal, save the settings with holding the MENU button as usual. The modification of wiki will follow in next days.

Purpose of this change is to enable new functionality to the rig, including resizing the waterfall/spectrum "on the fly" and more like virtual keyboard... So please be patient and report issues here, we'll work on it.
BTW the zeroing the least significant digits of frequency display works from now by taping on it (removed from the TUNE button).

73 Slawek

sp9bsl, 2018-01-23:
Hi,
since 2.7.65 we have ability to change the ratio of Spectrum to Waterfall display. To enter this mode please press and hold the graticule (frequency bar) and select new position for it, then press |SAVE|.
To store it permanently, do it as usual: long MENU press. Wiki change will follow. Have fun.

73 Slawek

DL8EBD, 2018-01-23:
Hello Sławek,
thanks for that nice feature, I test it this afternoon

df8oe, 2018-02-06:
This has been described in GitHub WIKI and amateurfunk-sulingen WIKI now and can be closed.

closed 2018-02-06 by sp9bsl