df8oe/UHSDR #1313: Spurs when xlate is ON

sh123, 2017-12-28:
Hello, 

I'm new to MCHF, latest HW, device is calibrated according to the tutorial, using it with firmware version 2.6.0, when XLATE is ON there are spurious emissions present and especially strong signal at -12 kHz XLATE. Is it something related to the firmware or some construction mistake?

![MCHF]
(https://i62.servimg.com/u/f62/19/62/65/46/spurio10.png)

df8oe, 2017-12-28:
-12KHz seems to be the LO frequency. This supr cannot be suppressed by firmware - only by hardware. Normally it is suppressed 50dB or better. If so: everything is fine. If not: hardware issue...

closed 2018-01-30 by df8oe

sh123, 2018-05-21:
It turned out to be excessive bias on the finals. After proper additional tuning all is good enough.