df8oe/UHSDR #1315: Slow Scan receive filter and bandwidth functions would be great.

WM0I, 2017-12-29:
I think all that would be needed would be filtering adjustments and bandwidth settings.

No need for the display to decode the slow scan. There are a few great decoders already for window and android computer and tablets.
Best  regards
WM0I

DG9BFC, 2017-12-31:
filtering adjustment (free setable by user) is not easy .... but adding some more precalculated filters could do the trick ... as far as i remember one of the programmers is/was working on this .. but maybe he has something different in his view right now ... 
he promised to add more filters ... so just wait a bit :-)

df8oe, 2018-01-03:
Filtering needs horse power. The better the filter is designed the more horse power it needs. So in my view the only way to go is:
- implementing a base set of filters which are neccessary for "standalone working" covering all implemented (!!) digital modes
- leaving "better" filters to the PCs, tablets and smartphones which are connected to radio

"Great decoders" normally do have their own implementation of very special filtering - no need for adding something for external programs to UHSDR.

closed 2018-01-03 by WM0I