df8oe/UHSDR #1317: Screen calibration problems

g7jiq, 2017-12-30:
This is what I have found when I calibrate my screen. (I have a standard mchf with a stock screen and using a stylus for the test)

I find V2.7.52 works fine, good accuracy and instant response when in cal mode and touching the cross hairs.
Drawing test is also very good. 
When running the radio, the magnify X* function touch position is spot on, as so is the other function positions. (these were previously slightly off)

When running the latest V2.7.53 I find several problems,
The calibration cross hair touch is not very reponsive and has to be touched several times to get a responce, and laggy between touches.
When the drawing test is run at the end I find I get only the odd white dot upon touching the screen, no line as before. and no response from the bottom right area.
When running the radio I find the screen beeps when touched (as normal) but is unresponsive or is way off the position it should be.
I repeated this several times and got the same results.

Rolling back to V2.7.52 and running thought the calibration corrected all problems, 

Hope this is of help....

sp9bsl, 2017-12-30:
Hi,
yes we're working on it. Stay tuned ;)

sp9bsl, 2017-12-30:
Because of no profit from the new version and problems wih the touchscreen we reverted the changes.
BTW Thank you for the report! The issue can be closed now.

73
Slawek

g7jiq, 2017-12-31:
I can confirm that all is working ok, and calibration is spot on.

Thanks for your hard work with the software,

closed 2017-12-31 by g7jiq