df8oe/UHSDR #1319: Option to flip the menu navigation for side-mounted encoders?

frazierjason, 2017-12-31:
The mcHF v0.7 board moves the general purpose encoders to the side of the unit, which presents a point-of-view issue for navigating the menus and changing values.  On all previous UHSDR builds, these encoders are oriented with the shaft pointing at the user, which turning clockwise would increase a numeric value, or navigate "down" in a menu.  But with the shaft parallel to the user's torso and pointed toward the right, the encoders now have an experience of rotating clockwise ("up") to increment values which makes sense, but scrolling in the menus it instead performs a down navigation event.  Can this be fixed, and if so, is it best done in a mcHF v0.7 specific build, or would a new configuration option be the best way?

df8oe, 2017-12-31:
Yes - this question is very interesting. New orientation of encoders do make different directions mandatory. Because of hardware of mcHF 0.7 and former versions is not firmware detectable (all the same...) the best would be a menu selectable way. We are not sure if it will be an "one config entry point" (like "mchf-version") or many config entry points (like "enc1 dir", "enc2 dir"). But this is on our todo list.

vy 73
Andreas

frazierjason, 2018-01-24:
I noticed one difference in the v0.7 schematic that was not present in v0.6:  a 4.7k pullup resistor on I/O pin PB2 (BOOT0) that goes to the BAND+ button situated across the anti-spark capacitor.  Were other versions prior to v0.6 also lacking that 4.7k pullup resistor?

I wonder if we could do any kind of trick to detect that pullup resistor, which would tell us we're on a v0.7.

df8oe, 2018-01-24:
This can be used to identify 0.7 boards. During boot we can add a pulldown resistor to BOOT0 and check if level at PB2 is low. If it is, we are on older revisions, if not, we are on 0.7. Possible issue can be that external resistor is to high so that level does not increase enough - but simple replacing (e.g. with 1K) would solve that.

df8oe, 2018-01-24:
But again I do not completely understand what is the issue (I do not have access to 0.7). Should directions of all encoders swapped at 0.7 boards? Or only some functions (would be much more different)?

frazierjason, 2018-01-24:
The encoder direction swap only needs to occur for v0.7 on any encoder action that visually correlates to movement of a visually vertical UI element.

For Enc2 this means logic reversal in UiMenu_RenderChangeItem(), to ensure menu item selection travels up and down corresponding to the twisting of the Enc2 on the right hand side.  

Enc3 only has a visually vertical effect when expanding a menu top group.  For example, twisting Enc3 clockwise ("up") while Standard Menu group is selected causes the menu items to drop "down" underneath the selection, and anticlockwise ("down") hides them upward.  To fix this, we would not change UiMenu_RenderChangeItemValue(), but instead UiMenu_UpdateMenuEntry() would need to invert the evaluation of ts.menu_var only for case MENU_GROUP.

Other Enc2 and Enc3 functions appear unimpacted because their feedback comes as an in-place value change, or an audio effect, and twisting the knob clockwise ("up") already works intuitively.

I will make these changes today on my own fork (frazierjason) and try them out.  For these changes ot work correctly in the main branch, we need to identify any radio with Enc2 and Enc3 knobs mounted perpendicularly on the right, bottom, or rear face of the radio.  At this time, it's only mcHF v0.7, which is why I mentioned the BOOT0 pullup resistor's presence.  Best to find a way to detect that resistor as-is, because there are probably too many kits out by now to ask everyone to make a change.  Also, it can't be converted to pull-down because that'd keep it perpetually stuck in DFU boot on power up.

frazierjason, 2018-01-25:
My two proposed changes in ui_menu_internal.c worked to improve vertical menu scrolling on v0.7 mcHF.  I pushed the simple changes into my fork from active-devel.

These alone aren't good enough, we still need:
- A way to detect we're on a side-encoders platform
- A place to store that detection (looks like SRAM2 byte signatures are used for other boot-time detections but no formal structure for SRAM2 usage is documented)
- Then to just read that SRAM2 signature in ui_menu_internal.c and switch on the UI menu navigation changes.

frazierjason, 2018-02-10:
Implemented in merged pull #1433.

closed 2018-02-10 by frazierjason