df8oe/UHSDR #1320: Ver 7 display

WD8BXS, 2018-01-03:
Will all of the latest firmware work on the v7 units, as it does not have SPI display?

sp9bsl, 2018-01-03:
Hi,
yes. Noone complained as yet. This is the same interface like in the OVI40 board (same 3.2" display).

73 Slawek

df8oe, 2018-01-03:
A short view on schematics does not show any difference in signal connections. So I do not see any reason why it should not work.

73
Andreas

hazzer23, 2018-01-03:
Any thoughts on how the front panel buttons will work betweenversion 0.6 and 0.7?
0.6 supports 17 front panel buttons, and the tuning encoder push button makes 18.
0.7 seems to only have 14 front panel buttons and an encoder push button, so Im assuming 15 in total.

73
Harry

db4ple, 2018-01-03:
The UHSDR firmware right now supports 17 buttons and 4 encoders as in the original mcHF layout.
No matter how many buttons you see, right now the 17 supported button connections have fixed meaning as seen in the mcHF 0.4 - 0.6 button layout. You should check the 0.7 schematic and probably you'll find that the 3 side encoders also have push buttons internally and are connected to M1 to M3 GPIOs.

73
Danilo

WD8BXS, 2018-01-03:
Well, Andreas, that sounds nice, but I have no idea what you said! 

Sorry,
Chuck

From: db4ple 
Sent: Wednesday, January 03, 2018 9:26 AM
To: df8oe/UHSDR 
Cc: WD8BXS ; Author 
Subject: Re: [df8oe/UHSDR] Ver 7 display (#1320)

The UHSDR firmware right now supports 17 buttons and 4 encoders as in the original mcHF layout.
No matter how many buttons you see, right now the 17 supported button connections have fixed meaning as seen in the mcHF 0.4 - 0.6 button layout. You should check the 0.7 schematic and probably you'll find that the 3 side encoders also have push buttons internally and are connected to M1 to M3 GPIOs.

73
Danilo

db4ple, 2018-01-03:
Hi Chuck,
this comment was a response to the comment made by Harry, hazzer23 regarding buttons, not your original question regarding displays. Harry should  have asked this in a separate issue or in the available forums and we would not have confused you.

73
Danilo

WD8BXS, 2018-01-03:
Thank you, sorry about my other reply to the post on the buttons, as I was confused...hihi

Chuck

WD8BXS, 2018-01-03:
Roger that, It had my head spinning, LOL,,,Have a Great New Year my Friend,

de Chuck WD8BXS

DG9BFC, 2018-01-04:
so issue can be closed???

WD8BXS, 2018-01-05:
Yes, thank You all!!

closed 2018-01-05 by WD8BXS