df8oe/UHSDR #1322: Red KED has tooo many functions

DG9BFC, 2018-01-04:
if we use noise reduction (threshold is controlled via red led) plus cw decoder (also on red led) we have two functions on same led on receive (plus tx led) .... should we not move one of the two to the green led?!?
ok then in cw mode both led would flicker ... but that way we could finetune cw decoder AND noise reduction .... now we have both on red led so you can only set noise reduction without cw decoder ... or cw decoder without noise reduction (or it is difficult to find the right setpoint)

df8oe, 2018-01-06:
Red LED is only needed for developing purposes.

Please ignore it if you don't like it.

DG9BFC, 2018-01-07:
i know that the led is for developing ... but it is difficult to use if you have two functions on same led (cw decoder thershold and nr vad threshold) ... thats why i asked to move one of the two to the green led (also only for development not for permanent usage) ... then setting nr and setting cw decoder could be finetuned for perfect (or near perfect ) decode

df8oe, 2018-01-07:
It is recommended NOT to activate NR during CW - so no impact of both functions.

closed 2018-01-30 by df8oe