df8oe/UHSDR #1326: Load 2.7.55 for testing. Here are the results I found.

WM0I, 2018-01-06:
Before loading the new firmware I went to the <>Code section to read what all has been changed or has been added.  There is a lot there to understand. Several years ago I programed in C++ for my job. Documentation was mandatory before releasing the software for production.

Now here is what I found when I first turned the radio on.

1. There was no audio. I had to turn the radio off and back on to get audio.
2.  Here again the red TX led pulsed on and off with the audio pulsing in time with the red TX led.
3.  When pressing the menu button the red TX led went out and the audio pulsating stopped.
4. I turned the menu function off and the audio and red led started pulsating again. I pressed the DSP button to take the radio out of NR + Notch. The audio and red led problem went away. I found that A+Notch and M+Notch filtering dose not work to kill the carriers.
5. I went back to the menu functions and selected the DEBUG to make adjustments which did not help in get the notch filter working.

I went back to the <>Code to see if I missed something I was to set but I couldn't see anything I missed.

At this point I went back to version 2.7.31. This version works the beat for my clone radio. Although I still get audio muting sometimes when I let up on the PTT. Then I have restart the radio to get the audio back.

Best regards
WM0I

df8oe, 2018-01-06:
Hi,

your observations about red LED:
This is actually used for testing new NR appoach which is under heavy development. Red LED shows  "speech recognizing" and flickers - that is wanted - we need that for development.

Your observations about manual peek and notch and autonotch:
These are not available at the moment because they would conflict with the development of new NR where we need reports of most as possible users. If all say "we stay at xyz until it is finisehd" it NEVER will be finished :) Notew: we are not a commercial institution. We do not have employees that are doing the testing job. And we do not have time to write documentations for each change that was made. If you want to be "up-2-date" you have to dig eround the GitHub commits and follow the discussions of our UHSDR team at https://www.amateurfunk-sulingen.de/forum/index.php?board=19. If you want to post there: you are welcome. Sometimes you will need Google translate because of discussions can be in English or German - but Google does a great job. Only if you follow both (!!) you do have all informations you need.

Your observations about "no audio":
We cannot confirm or reproduce that. There are one or two other OMs with same issue but the big mass does not have. We are not sure if it is a firmware issue - mainly we think it is a hardware issue.

WM0I, 2018-01-06:
I understand about the documentation being an overhead task. 

For the benefit of those who don't read C++ well or not at all I think a small description of the change/ added/ deleted would help others with their testing.

Just a thought. I am find with reading the <>Code.

Best regards
WM0I

df8oe, 2018-01-06:
Hi Rich,

we already do that in our commit comments. Sometimes - especially when things are moving extensively - this is not possible or the information would be outdated at the next commit (but writing does take 10 minutes). Other hand: if you do not read all commit comments maybe you mised one where the information you are looking for is already present - we cannot add it to every new commit. So it is always a good idea to discuss "special behaviour" as the one with the red LED in our discussion group. Motst changes are never recognized or mentioned by most of the users. But if someone has remarks or questions maybe much more do have the same questions. So we can react to such questions in our forum.

We are doing this in our forum and not in the Yahoo group because UHSDR is not only related to mcHF and mcHF is a "restricted group".

vy 73
Andreas

DG9BFC, 2018-01-06:
if you want a working notch you have to step back to version 2.7.31 (with old notch filter)
all behind that are developing the NEW NOISE REDUCTION ... a long tone (beat note and carrier) surpression is added ... but it is not as good as manual notch or the old autonotch
you can best help the programmers if you give them some reports about the noise reduction (and they sure will add a good autonotch soon!!)
when the new notch will be only 10% as good as the new noise reduction ... then it will be perfect ... but it may take some time :-) ... wink wink well done boys (test reports from here will follow soon)

WM0I, 2018-01-06:
I have switch back to 2.7.31 and after putting all the settings back it is again sounding like FM radio ( almost, Hi hi ) When in CW mode all I have to do is move out of NR+NOTCH @ a setting of 10 to the DSP setting of NR of a setting of 10.  Would be nice when going into CW mode that the NOTCH part be disabled. Then when back in SSB mode the NOTCH would be enabled again.

Just thinking.

Best regards
WM0I

closed 2018-01-07 by WM0I