df8oe/UHSDR #1328: How to modify an older Version of firmware?

WM0I, 2018-01-07:
If I need to make small modification to 2.7.31 how would I go about it?

I would like to disable the notch filter when I select CW mode.
I would like for the notch filter to be enabled when in all the other modes.

How do I get all I need to do this?

WM0I

df8oe, 2018-01-07:
Hi Rich,

"git" is a command line versioning system. Because of I am Linux user the command line is a mainly used thing. There are some GUI approaches (like "gitkraken") but they eat much horse power and cannot reach the power of CLI commands. In short:

1) install "git" (on a Linux-System this is included in each standard installation)
2) open a console and cd <into_your_directory_you_want_to_work>
3) type git clone git@github.com:df8oe/UHSDR.git
4) after download is complete you have access to the sorce code of EVERY commit that ever exists in history. To change to a specific commit simply type git checkout <commit_id> and your folders contain the files of that commit.
5) to return to recent stage type git checkout active-devel

That is "git in 20 lines" ;) Of course there are much more git commands and many of them can carry parameters, too. But for starting this is enough.

Good luck
Andreas

db4ple, 2018-01-07:
Hi,
of course you also need to setup a development environment (see our Wiki with detailled instructions).
A hint for git: To make your own changes in a proper way to an existing commit (read version), you should create a branch in git which starts with exactly this commit. You do this after the git checkout (as mentioned by Andreas, you do `git branch new_branch_name_goes_here`. This will make it possible to later commit these changes to your public github (of course this is optional). 

However, please note, that in general for obvious reasons we do not accept changes  for older versions for inclusion in the active development.

DG9BFC, 2018-01-07:
as far as i could test the autonotch is not active in cw mode (in ver 2.7.31)... so what do you wanna change?!? .... is see you miss that autonotch ...(me too in some rare cases)
the programmers are working on the nr now ... autonotch will come soon!! they promised that (or will bring back the old autonotch) ... just wait or help them with testing new version (we are now at 2.7.58!!)

DG9BFC, 2018-01-07:
correction ... now at 2.7.60 :)

remember ... the programmers do need test reports ... otherwise they can not finish nr (and so no time to add notch)

closed 2018-01-12 by WM0I