df8oe/UHSDR #1340: Still getting audio muting.

WM0I, 2018-01-15:
Actually there are 2 conditions which cause this problem. 
1. After I stop transmitting and let up on the ptt the audio goes into very loud audio  and it takes several seconds to quite back down. Or after I let up on the ptt I get audio muting.

2.  Sometimes when I turn on the transceiver it will come up muted.

When I get the muting I have to reboot the transceiver to get the audio back.
Very frustrating to have to reboot and miss the conversation.  

Also I was not able to change the display from SPI to Parallel. The display is soldered in. 

Best regards
WM0I

df8oe, 2018-01-15:
Because of very rare reports about "audio muting" I am nearly sure this is a hardware issue. But I don't have any idea where to start investigations. It can be dirty power supply of MCU or of audio codec, missing or bad placed blocking capacitors on Vcc, missing or wrong value of serial resistors in I2C or I2S lines, wrong zenerdiodes on paddle lines - and hundred more possibilities...

Display soldered? No comment. A result of the "nice small case". In my eyes not a good choice for this radio. I suggested in spring 2015 to put LCD on headers ;)

73
Andreas

WM0I, 2018-01-16:
Well I have decided to sell my clone, so I will still help testing on the version of firmware untill I no longer have the radio.

77's
WM0I

Tibet4690, 2018-01-16:
A little off topic but mind if I ask what you got for it second hand?

WM0I, 2018-01-17:
I don't mind you asking. I haven't sold it yet.I am asking 275.00 I paid 450.00 for it on ebay last September. I got it for RV'ing but my health is going to keep me from traveling. Even with the audio muting I had great fun out in the wild making QRP contacts. Since I am home bound now I need a big radio.
Best regards
WM0I

closed 2018-01-17 by WM0I