df8oe/UHSDR #1342: Request to emulate FT-817 control of remote T1 ATU.

RightRudder, 2018-01-20:
It would be very nice if the USB host port on mcHF could control the Elecraft T1 tuner in the way that the FT-817 can.

df8oe, 2018-01-21:
If one of the developers has time AND such a tuner for testing AND there are more who wishes this feature maybe it will come up one day...
Or, second solution: someone else wants to contribute it :)

vy 73
Andreas

lb5va, 2018-01-29:
Aha.. This is a really good idea! It's a pitty that not more people encurraged this. The T1 is a very nice and small tuner. Perfect in combination to the mcHF. But It normally changes band automaticly without.
But - there is PA's that use this function. Tokyo Hy-Power smaller PA's uses Ft-817 interface.
Stig - LB5VA

RightRudder, 2018-01-29:
@IB5VA the T1 only retunes when the button is pressed or it gets the band change info from the remote interface.  The button press requires that you transmit on the new QRG in order for it to tune.  Getting the band info through the remote interface, it switches to the L/C values last used on that band so you don't have to cause QRM on the band.  I hope the dev's will decide this is slightly more than a "very low priority"  haha

df8oe, 2018-01-30:
Priority will not change because these things have not changed:

- none of the developers do own such a tuner
- none of the programmers actually do have time for this
- there are only two who asks for this feature

lb5va, 2018-01-30:
I can understand this. :-)  Thanks for all the other good work!  Stig

EA2OY, 2018-01-31:
Hi, another one with Elecraft T1 Tuner, i miss this feature in McHF, i think that its easy to implement, t1 works with 4bit signals for each band and tune button. but I understand that the programmers of mchf doesnt have that tuner. Otherwise. THANKS A LOT FOR YOUR HARD WORK ! 73 de EA2OY Jesus

RightRudder, 2018-01-31:
Hello Jesus

Thanks for adding your name to this request.  Please be aware that if you build the arduino dongle from the link in the first post of this issue, then it is thye dongle, not mcHF which takes care of the 4 bit control to the TI.  All McHF has to do is output an ascii string wiith the band data and state of tune button.

EA2OY, 2018-01-31:
i understand that if mchf outputs this string, i can connect with arduino and t1 to work like the 817 really?

I hope that can be put on mchf firmware... thanks !

df8oe, 2018-02-02:
I have reviewed the links and can state following:

1) firmware acts mcHF as a "USB CLIENT". So it is *mandatory* to plug a USB HOST to it ant this can talk to it via a "virtual comport". There are *no free GPIOs on mcHF* to add a real comport.

2) OVI40 does have enough free GPIOs (and some of them can be used as serial port). So definitely this will be added on OVI40 - but not on mcHF.

closed 2018-02-02 by df8oe