df8oe/UHSDR #1343: Well I unbricked my Clone!  Thanks for all the advice!

WM0I, 2018-01-21:
The most of the problems came from bad solder joints. and I had to solder in wire jumpers in a few spots. I now have better audio. Just about all the noise is gone and and have now a very low noise floor. I also have control of the Power out after I did the calibration.  I would say now to all those who are having trouble with the clones is take it apart and clean the boards and go over the solder joints that look frosted and gray. Now I hope there is no audio muting. I am now testing that now. 

Best regards
WM0I

df8oe, 2018-01-21:
Hi Rich,

thank you for your informations about the technical quality of the clones. I think that will help others to point their searching for issue solutions to the right direction.

73
Andreas

df8oe, 2018-01-28:
can be closed now.

closed 2018-01-28 by df8oe