df8oe/UHSDR #1346: What I found about audio muting on my RS-918ssb clone

WM0I, 2018-01-21:
I was running test on the audio being switched from TX to RX muting. It seems that if I clicked the PTT at times, not all the time but maybe 1 out of 15 to 20 times it will mute. If I speed click the PTT it will mute sooner. I went looking for anything in the menu's that would cause a software racing. Not seeing I went looking at the menu information. I found something about bus speed.  I set both bus speeds to 60khz. Doing the same test as before I got no audio muting and it didn't seem to hurt anything else.

Firmware version  2.7.31 

Hope this helps.

WM0I

{ MENU_CONF, MENU_ITEM, CONFIG_I2C1_SPEED, NULL,"I2C1 Bus Speed", UiMenuDesc("Sets speed of the I2C1 bus (Si570 oscillator and MCP9801 temperature sensor). Higher speeds provide quicker RX/TX switching but may also cause tuning issues (red digits). Be careful with speeds above 200 kHz.") },
    { MENU_CONF, MENU_ITEM, CONFIG_I2C2_SPEED, NULL,"I2C2 Bus Speed", UiMenuDesc("Sets speed of the I2C2 bus (Audio Codec and I2C EEPROM). Higher speeds provide quicker RX/TX switching, configuration save and power off. Speeds above 200 kHz are not recommended for unmodified mcHF. Many modified mcHF seem to run with 300kHz without problems.") },

df8oe, 2018-01-22:
60KHz?? That is very, very slow!! My mcHFs all run at 320KHz - flawlessly.

Reasons can be:
- They cloned the 0.4-issue (fit C38 & C39 on rf board)
- They have made the PCB design much worser so that I2C bus is hurt
- They use WM8731 clones chips which do have very low I2C speed limits

Every mcHF which is built from Chris PCBs runs absolutely stable at the default speed of 10KHz, most do run at much higher rates (than mine) of 320KHz.

WM0I, 2018-01-22:
I caused Problems when I tried to fix some bad solder joints before and I had to make wire jumpers to fix them. So I am going to leave well enough alone unless reducing the speed cause issue then if I have to I will find someone to make the changes for me. From what I am seeing now all is OK.

Best regards
WM0I

DG9BFC, 2018-01-22:
check if the capacitors c38 c39 are on the board
my mchf runs fine with 340kc bus speed
so andreas is right ... they cloned the 04 design and made the fault with the "filter caps"

df8oe, 2018-01-30:
this is not a firmware related issue...

closed 2018-01-30 by df8oe