df8oe/UHSDR #1347: Free DV 700c Mode?

ei2gyb, 2018-01-22:
Are there any plans to add the FreeDV 700c Mode to the radio?

Steve EI2GYB

df8oe, 2018-01-22:
Yes we will add it. But it will not work on mcHF due to MCU lack. But on OVI40 it will work :)

db4ple, 2018-01-22:
Hi,
btw, I don't think it is much more RAM for 700c but for sure more cpu power. That is why the STM32F4 is very likely to be not sufficient for 700c. 

73
Danilo

DG9BFC, 2018-04-27:
david rowe is working on a 700 d mode
he wrote ... low cpu power and low ram needed ... he just needs some "coders" to work with him
maybe we can get the 700d mode inside mchf'?!?

db4ple, 2018-04-27:
I am pretty sure we can. At least in the OVI40 there should be no issue. However, low ram and low power are fairly relative. For the mcHF anything with more ram or performance requirement, even slightly more. is almost too much.

DG9BFC, 2018-04-27:
if the 700 d mode does not need more as the 1600 mode it should be possible

df8oe, 2018-04-29:
If you are satisfied with the informations: please close this issue. Thank you!

closed 2018-06-09 by df8oe