df8oe/UHSDR #1348: please add a shift in the display for zoom ratio x32

DG9BFC, 2018-01-22:
title says it all
zoom ratio x32 shifts the signal out of the view so that high zoom ratio can not be used for precise tuning on a psk or rtty signal
with a shift in the view area (so that usb or lsb filter are both centered in x32) we could use the better view for exacter tuning on a signal
(i dont know if the bpsk receiving routine can give a frequency readout towards the tuning ... that woud be a "snap function" in bpsk and another nice add-on and could be added later)

phaethon, 2018-02-09:
As shift frequency for bpsk is now changed to 500 Hz, x32 zoom is not outside the screen. So, the original issue is not relevant for bpsk. On the other hand seems that x32 is not enough (at least for me). There was an idea written elsewhere (in commit or PR comments I think) that we could do a split screen with a close-up for a small part of the spectrum (e.g. BPSK shift frequency in this case).

DG9BFC, 2018-02-09:
it is not for 500hz bpsk but it still is shifted half of the filter outside window
the reason for asking was not only because the 1000hz bpsk outside of the view
in my view the shift would be useful for ANY mode not only for bpsk
in higher zoom ratio i wanna see the complete filter .. and not half of my filter and half of the unused sideband
there is already a shift from x1 to x2 and above (shifts the carrier point inside centre
so why not addan additional 3 kc (plus or minus) ith higher zoom ratio (up or down depending on sideband used)
so please let that issue open ... it was NOT only for bpsk it is for ALL modes in zoom ratio x32

sp9bsl, 2018-02-10:
Hi,
yes there was an idea to make separate view for zoomed receiving bandwidh, but I think this is for 480x320 resolution and F7 core (OVI40 board). In F4 core there is not enough RAM to do this, beacuse we need RAM for another quite big FFT and spectrum display data. Also the horsepower of F4 would be a problem (remember NR+NOTCH helicopter issue?). This feature is on our TODO list anyway :-)