df8oe/UHSDR #1354: The height of the waterfall is not recorded in memory

ly2bok, 2018-01-23:
In ver.2.7.66
After setting a new waterfall height, the value is not stored in memory. Remains in the middle. SDR mcHF ver.04. 
Justas LY2BOK

df8oe, 2018-01-23:
Hmmmmm. Here it is....  Have you used the "SAVE" option at touchscreen when sliding the positions?

ly2bok, 2018-01-23:
Yes, it is clear to me how to write it into memory, but it just does not work ... After recording, 
waterfall or spectrum on the screen. If we go to both, we get again two even parts.

sp9bsl, 2018-01-23:
Hi Justas,
when you select the new position, you have to exit to "normal" work by pressing SAVE, you do right.
After exit you have MENU button marked that you should save the values to eeprom (long press MENU as usual). But (!) if you change the view to one of three presets WATERFALL/SCOPE/DUAL (menu item or bar selection) it resets your earlier settings. To go back, simply restart your rig.

73 Slawek

ly2bok, 2018-01-23:
https://youtu.be/GjZauOSJjiY
Good evening, Slavik.
Where am I doing not right thing? It's funny, but does not write in memory. Why such a "complex" function?

sp9bsl, 2018-01-23:
Hi,
thank you for the video. You just do exactly what I said above. By pressing the BND+ and BND- you select the predefined setup 0%, 50% or 100% of height for freq bar location (like it was before the 2.7.66). 
When you select new location by long press the freq bar you change the location to your favourite between 0% and 100%, then save it by long press the MENU or POWER. It will be at this point when you unplug the power supply and plug again or restart until you use the BND+/- again.

Looks like you need the another preset (fourth) which will keep your favourite setting to be able to select it with the BND+ and -.

ly2bok, 2018-01-23:
My SDR mcHF ver 04 does not work this way. Or a software error, or I'm a complete fool ... Although I've been working with computers since 1986 ..... I've been trying 3 hours .... Hi  hi...

df8oe, 2018-01-23:
Try following:

1) set height in dual mode as you already did. Enter with "SAVE".
2) long press MENU (for storing to EEPROM)

**Now - very important - do not touch BAND* and BAND- simultaneously or change from dual mode to any other (this would definitely destroy your percenteage settings!!)**

Power down mcHF

Power up mcHF -and - voila! percentage settings are present again!

But if you switch from dual mode to another YOU LOOSE PERCENTAGE.

ly2bok, 2018-01-24:
Yes ... it turned out that there was an error in the program. For example, I can not make 10% waterfall and 90% of the spectrum. It is recorded in memory as 100% Spectrum. The same with a 20% waterfall. In memory, you can record only about 30% or more of the waterfall. This is the problem, why I could not write the settings in memories.

sp9bsl, 2018-01-24:
Hi,
right, this is intented. What you will see on scope with 10% of its height? Same wor waterfall. it filters out the settings if above 16 pixels to top, and below 16 pixels to bottom.

df8oe, 2018-01-24:
:) I have waited for this comment.... For me it was and is absolute senseless to choose such small areas so I never would have tried...

Percentages should be in a range that makes sense.

DD4WH, 2018-01-24:
I am adjusting the waterfall height to about 35%, press SAVE, save the settings by pressing the menu button.
Then I change between scope, dual and waterfall as usual by pressing buttons band+ and band- simultaneously, but the dual view will then not take the saved 35% setting, but always uses 50% spectrum/50% waterfall.
is that a bug or intended behaviour?

DL8EBD, 2018-01-24:
Frank, siehe Andreas posting weiter oben...
erst einmal Neustarten, dann Band+ und - benutzen...

_Now - very important - do not touch BAND and BAND- simultaneously or change from dual mode to any other (this would definitely destroy your percenteage settings!!)*

Power down mcHF

Power up mcHF -and - voila! percentage settings are present again!_

DD4WH, 2018-01-24:
I save the settings (35%), then save with menu button, then I power down, power up.
And if I now change the view to spectrum and change to dual again (with band+ & band-), I expect to see 35% again.
So this switching back to 50%, although I have saved another setting is intended behaviour?

sp9bsl, 2018-01-24:
Frank, you save the current setting. If you set it to 50%, you will save it as you set.

DL8EBD, 2018-01-24:
Slawek, maybe a 4th setting behind BAND +/ -
1. Spectrum only
2. Waterfall only
3. Splitscreen 50/50
4. Splitscreen user

sp9bsl, 2018-01-24:
Thomas yes, but from firmware point of view the number of selection is a combination of two display flags: one for waterfall and one for scope. I have to add a variable. Just need some time for analyze :-)

DL8EBD, 2018-01-24:
Slawek, I think it has a low priority.
there are more important construction sites

df8oe, 2018-01-24:
I think Slawek has proposed the working condition like mine:
If a user wants to use dual screen and does not like 50% setting he can change it for his purposes. And then he never changes the view to scope or waterfall. That is exactly my working condition and all works as expected.

ly2bok, 2018-01-24:
OK, I understood how Slavik conceived. The fourth paragraph is "Splitscreen user" OK!
It is necessary to finish each function to the end. A lot of uncompleted functions in SDRmcHF. It is possible to think up much, but it is necessary that it worked.
Good luck. Thank you for your work. Justas

closed 2018-01-24 by ly2bok

df8oe, 2018-01-24:
It is a nature of "Open Source" that it is NEVER finished and ALWAYS things are incomplete... That means exploring and developing - amateur radio is an experimental hobby.

DD4WH, 2018-01-24:
Hi Slawek and all,
thanks for all your comments and contributions. I think I have now understood the way of thinking:
many of you set the display once and then never change it.
From my point of view, I change the display quite often: waterfall, spectrum or even both (since Slawek invented that great feature!). And whenever I change to dual view, it would be nice to have the saved proportion spectrum/waterfall restored :-).
But I agree, this has low priority.
And I also agree regarding the "finished" product UHSDR. It simply does not exist, everything is in a flow and we are all trying to optimize and extend things all the time. Yes, you can call that incomplete ;-).

sp9bsl, 2018-01-24:
Hi Frank,
I completelly agree! You guys gave me precious opinion on how it should work. My opinion is that unfinished thing sooner or later will go back in mostly unexpected moment... So, I will finish it before proceed to the next step in our rig. Danilo prepared code that is quite easy to change, it only requires surgeon knife to do it right :) Stay tuned!

sp9bsl, 2018-01-24:
I added the storing way as you suggested. It is simply DUAL view mode, but now it is "customizable" (no fourth selection). Also I added the possibility to leave the selection by pressing the **X** or accept new selection by pressing **OK**. After this, simply save to eeprom. Now you can change the view by pressing BND+ and -   or selecting from upper bar or selecting via menu. Setting for DUAL view stays where it was set. Upper margin is 24 pixels (minimum scope size without bar, with bar is 8 pixels) and lower is 16 pixels (minimum waterfall size).

73 
Slawek

DL8EBD, 2018-01-25:
thx to Slawek, I think it's a resonable compromise for all user
73
Thomas