df8oe/UHSDR #1355: unable to change spectrum / waterfall view using band switches in 2.6.77

hazzer23, 2018-01-23:
Previously I could toggle the display between waterfall, spectrum, and dual, by holding down the  band + and - switches - this no longer seems to be working in version 2.7.66 - I had found it to be a useful shortcut.

73
Harry
MM0MHP

sp9bsl, 2018-01-23:
Hi Harry,
this is new knowledge to me. I see the problem.

73 Slawek

ei2gyb, 2018-01-23:
I had this problem after 2.7.66 after the update 
however i went into the menu and saved a change and it worked fine since then.

73 Steve EI2GYB

df8oe, 2018-01-23:
Hmmmmm. Very interesting. I didn't know this feature, too. One of my mcHFs do have 1.7.64 and I tried it - but long press on BAND buttons didn't change anything. I never used BAND buttons for changing WF/scope and I cannot confirm it works on 2.7.64...

DL8EBD, 2018-01-23:
Andreas, both buttons together!
Until D2.7.66 it was available and I used it periodical to switch between scope, waterfall and splitscreen

hazzer23, 2018-01-23:
Update:
On my mcHF the "band button" short cut is working as long as the display has a title bar 
ie.... SC(7.5db/div)   < Magnify>)    
If the title bar is removed (with a long press on the title bar) the "band button" short cut stops working.

73
Harry

DL8EBD, 2018-01-23:
that is an mcHF property :-)
on OV I40 it definitively works without  "Magnify title bar" == Spectrum size "big" until yesterday

sp9bsl, 2018-01-23:
Yes, that right. I Added this to prevent accidentally change of waterfal/scope size when the bar is off (that can be if you touch small scope in area where the selection is localized on the upper bar. But i didn't know that there are shortcuts on BAND+/-. I see it in the code now.

sp9bsl, 2018-01-23:
Sometimes the description of commit takes longer than the change itself ;)
Anyway, should work now (tested and works on my rig).
Many thanks for testing!!

closed 2018-01-23 by df8oe