df8oe/UHSDR #1360: 2.7.68 test report. Not good report

WM0I, 2018-01-24:
High noise as soon as I turned on the new firmware. Red light on all the time even after cycling a couple of times. I opened up the DEBUG menu and made several adjustment even did the release. No help.

One step backwards. 
regards

WM0I

df8oe, 2018-01-25:
**I cannot confirm your report in any way.**

Hmmmmm..... Maybe not good test?

There has absolutely nothing changed that can influence the noise level.

Staying on "bleeding edge".

73
Andreas

hazzer23, 2018-01-25:
Im not seeing a higher noise floor on my mcHF using 2.7.69. If anything the noise floor is lower on the recent firmware releases (-104 dbm with no antenna attached).

73
Harry

RightRudder, 2018-01-26:
I can also confirm I have seen no difference in noise level with 2.7.69 FW on my v0.7 machine.

df8oe, 2018-01-27:
I will correct topic naming because it does not show reality.

closed 2018-01-30 by df8oe