df8oe/UHSDR #1362: SWR meter

finelld, 2018-01-26:
Is there a way to access the SWR meter function without the touchscreen?  I had to remove the resistors to allow the use of the touch screen to get my MCHF to complete the boot cycle.  In other words, I don't have touchscreen capability.

73's, 

N7LRY

hazzer23, 2018-01-26:
Repeated long presses of F2 will cycle the display meter through ALC, Audio and SWR.
73
Harry

DG9BFC, 2018-01-26:
????? you had to disable touch that mchf boots up??
you should not ask how to switch meter fuction without touch ... you should ask how to realive touch and let it STILL boots ... that would be the right way

ps there is a menue entry point for use without touchscreen

finelld, 2018-01-26:
> <reply+0222ad4f0de56f72b71c4244c62691a3617b3241105ab4b392cf000000011682e5599
> 2a169ce116469c8@reply.github.com>
> <author@noreply.github.com>

My MCHF was getting frozen on the splash screen during startup.  I was told
on this site that this is a less common issue but that I needed to remove
the resistors for the TP/LCD SPI Interface for the radio to complete its
boot.  This did solve the issue.  But, if I am not mistaken these are also
the resistors that enable the touch screen.  Please correct me if I am
wrong.  I have the v0.6 boards.and am running version UHSDR 2.7.39.  I was
not told what causes the anomaly with some builds resulting in these
resistors needing to be removed.

I have looked for a menu entry point for use without the touch screen and
have not been able to find it.  I have found some settings that do have menu
entry oints for use without the touch screen like the zoom on the
scope/waterfall, but not for the meter display.

Thank you for the response.

Best Regards,

David Finell, N7LRY

<https://github.com/df8oe/UHSDR/issues/1362#issuecomment-360775830> , or
mute the thread 
TyogG-aks5tOclZgaJpZM4Rt1Js> .

  
b","name":"GitHub"},"entity":{"external_key":"github/df8oe/UHSDR","title":"d
f8oe/UHSDR","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/df8oe/UHSDR"}},"updates":{"snippets":[{"ic
on":"PERSON","message":"@DG9BFC in #1362: ????? you had to disable touch
that mchf boots up??\r\nyou should not ask how to switch meter fuction
... that would be the right way\r\n\r\nps there is a menue entry point for
use without touchscreen"}],"action":{"name":"View
Issue","url":"https://github.com/df8oe/UHSDR/issues/1362#issuecomment-360775
830"}}}

df8oe, 2018-01-26:
R30...R32 must e removed if you do want to use your LCD driven in parallel. But that is completely independent from touchscreen which has its own SPI connections!

First you can look into system menu if touchscreen is already recognized. If so: you must calibrate it before use.

If not: lokk what resistors must be fit to put SPI for touchscreen on HY28B headers. You will find informations you need in our WIKI here on GitHub.

73
Andreas

closed 2018-01-30 by df8oe