df8oe/UHSDR #1363: Is saving 48 bytes worth the trouble?

frazierjason, 2018-01-26:
While reviewing some other stuff, I saw today that UHSDR binaries are including STMicro's USB Host mouse drivers.  It was mentioned that 512KB mcHF are running out of space, so I took a stab at removing the mouse support.  I modified usbh_hid.h/c to just ignore or fail away any mouse devices and not load the usbh_hid_mouse.h, and removed the mouse files from all the .mak files.  Everything builds fine, and the resulting .bin file is 48 bytes shorter.

Unfortunately it's not quite working, the mcHF immediately hangs whenever I plug in a composite keyboard/mouse device.  I don't have a pure generic keyboard that even the stock mcHF is happy with, only some MSFT keyboards that refuse to work.  I am happy to continue debugging, and I work with a guy with really solid HID background, so I think I could get it working.  But should I even bother?  Is 48 bytes worth the trouble?

df8oe, 2018-01-26:
48 Bytes definitely is NOT worth the trouble. And digging into HAL and STM stuff is not neccessary in most cases. Linker does not include functions that are not used. So you may win some milliseconds for building - but that is not worth the trouble, too.

I do not think there can anything be done against running out of flash for 512KB MCUs. You can delay the crossing of this border (some small) time - but you cannot prevent it.

It was known and pronounced from early 2015 on that it *is a very good idea* to fit 1MB MCU instead of 512KB. The costs are some Cent more. If you tell something and nothing happens maybe you lost your reliability :) So now this point has come - as pronounced in 2015. And if Chinese cloners fit 512KB MCUs on devices which are actually sold - it is their problem (and the problem of the ones who buy clones which are optimized for profit.

db4ple, 2018-01-28:
Also from my side: This is absolutely not worth it. Unless absolutely necessary please don't change the STM provided library code. Yes, I had to do this in order to have a composite driver for combined serial and audio in the UHSDR firmware, and thus I know what I am talking about.

closed 2018-01-30 by df8oe