df8oe/UHSDR #1364: mcHF v0.7 switch OFF problem

iw5bms, 2018-01-26:
Hi, I recently bought a last mcHF version and I still have  old problem about switch OFF.
I read about this problem in other post but I understand that this bug has been not solved
from both side HW and/or SW. I have not substitute firmware  and I'm using D2.105.106.
May be it can be solved using different firmware but I like to receive confirmation about this.
Thanks for help.
Maurizio iw5bms

df8oe, 2018-01-26:
There is recently no firmware solution and we do not know which hardware issue causes this. It is a hardware problem - that can be confirmed. But we do not know where it can be found. You can stay at lateste "bleeding edge" firmware (which is D2.7.69). Our old approach to solve this impacts RTC mod on mcHF so there is no solution. From a logic point of view it is impossible what you see. But we do not have one mcHF which has this issue and can take a look.

DD4WH, 2018-01-26:
Also, from my point of view:
- you bought hardware from a seller (mcHF 0.7 from Chris)
- you use the software provided by this seller
- it does not work

Then the logical first solution would be to ask the original seller (Chris) about this problem.
Just my two cents.
Frank DD4WH

iw5bms, 2018-01-26:
Good morning Frank, yes I bought mcHF from Chris in December and using the original sw by the Seller.
Sw works but not totally many items are critical but now I would like to solved this problem. I h've already ask Chris and he send me not satisfy answer because he believe that problem comes from mechanical part . My mcHF is still completely open and there is not mechanical problem from (S17 )push button.
Maurizio iw5bms

DG9BFC, 2018-01-26:
that is superb user support (sarcasm off)
did you already tried a newer firmware??

iw5bms, 2018-01-26:
Hi Siegfried, thank's for your suggestion. I will try to upload new firmware only as "last beach". I'm 99% sure that it's hardware problem. Chris explain me that CPU has internally pullups and pulldown resistors controlled by code, but if my case can be considered rare I don't believe that problem is the software.
Maurizio iw5bms

iw5bms, 2018-01-26:
Contrary at my previously message software was the origin of problem. I have been able to solve the problem using with bootloader v 0.14 as suggest by Chris. I thank everyone for their help and I hope that this experience could be useful for other builders who could have similar problem.

closed 2018-01-26 by iw5bms

DG9BFC, 2018-01-26:
0.14 is also old