df8oe/UHSDR #1369: Question, If upgrade the bootloader from 3.4.5 to 3.4.6 will I still be able to load 3.7.31?

WM0I, 2018-01-28:
I want to help the testing but I don't want to cause myself problems. 

WM0I

df8oe, 2018-01-28:
The only thing that has changed in bootloader is reimplementation of former anti-reboot-loop code. All who never have had reboot loop do not need to upgrade.

3.7.31 is outdated - we do not need any test reports for that version. Tests are welcome and appreciated for newest firmware.

73
Andreas

WM0I, 2018-01-28:
No I am not retesting 2.7.31.  What I do in testing is I load the new firmware such as 2.7.72 and do the testing when I finish with testing 2.7.72 I go back to what I consider is what really works well for my clone. I just wanted to make sure I can load 2.7.31 back again. 

And no, I do not have a rebooting problem.  

Andreas thanks for the heads up!

73
Richard WM0I

closed 2018-01-30 by df8oe