df8oe/UHSDR #1371: 2.7.72 loaded, 3.4.6 not loaded.

WM0I, 2018-01-28:
Since I was not having problems with reboot I did see a need to change bootloader from 3.4.5.
Powering the clone I went straight to the DEBUG menu selection and changed the relasse  to Devel2 and to my noise level came down a lot. Not as much as I hoped. But with a S9 noise level I was able to reduce the noise actually heard in the speaker to what I would call a S6 to 7. UP and DOWN noise audio level.

As far as the RTTY and FREETV and PSK functions in the code I was not able to test those. I have no PC software for those, nor did I have a small keyboard to test that section. CW code reception has always be good. I don't use the CW decoder and I can't seem to get the adjustments right from the time I bought the clone. Actually I have never use a encoder or decoder. I try to copy CW in my head only. 

Here again the noise level on 80/75 meters and 40 meters was down low enough to hear weaker station.

I really like the 2.7.72 audio that has less noise over the 2.7.31 witch has the NOTCH filter that works. 
Audio levels on 2.7.72 or 20% loader than 2.7.31 at the same speaker setting.
 And hear again an automatic NOTCH filter that works just like 2.7.31 would be great.

Looking forward to the next version.
73
Richard WM0I

sp9bsl, 2018-01-28:
Hi,
I just wonder: could we have reports in one issue?
Richard, don't get me wrong - your reports are important but we are flooded with new issues, it is hard to find the real problem between all your issues.
73 Slawek

WM0I, 2018-01-28:
I am sorry I thought I was only reporting on one issue.  I re-read my post and I see where it is unclear what I was I testing and what I couldn't test. The last part of the post was just a wish for a good automatic NOTCH filter. I will keep to the point of my testing.

73 Richard

closed 2018-01-30 by df8oe