df8oe/UHSDR #1377: OVI40: rounding off the hundred hertz digits does not work properly in CW Mode

DL8EBD, 2018-01-29:
If I have set a frequency, e.g. 7.097.123 Hz and touch on the "123", then the display jumps to 7,097,000 Hz
All good...

But not in CW!
There it jumps to 7.096.350 Hz

why not  to 7.097.000 Hz as usual?
Is this a bug?
Incidentally, it always jumps to x.xxx.350 Hz - no matter what was chosen for a QRG.

The mcHF with one of the last firmware makes it right

db4ple, 2018-01-29:
Hi,
without looking into the code, which I should have done: that is xxx.000 - 650 , I venture to guess you have 650 Hz as sidetone frequency.

DL8EBD, 2018-01-29:
yes Danilo... 
set it to 1000Hz and all is okay
thanks ;-)

DL8EBD, 2018-01-29:
STOP!
on mcHF I have 650Hz sidetone too.

df8oe, 2018-01-30:
We have had a discussion about rounding ~1...1,5 years ago. We agreed that in SSB it should be 1KHz - but in other modes it should be different. AM it could be larger, CW it could be smaller. This discussion was never ended successfully. So we first should discuss what grid should be used in other modes.

DL8EBD, 2018-01-30:
okay, but why is that between mcHF and OVI40 solved differently?

df8oe, 2018-01-30:
That may be due to the fact that lo on mcHF doeas work in complete other way than on OVI40. There it must not work on f x 4 (except you are running mcHF RF board...)

db4ple, 2018-01-30:
Hi,
it is not solved differently in the code. Please check the code before making too wild assumptions:
Below is the code from ui_driver.c which does the work:
```
static void UiAction_ChangeFrequencyToNextKhz()
{
	df.tune_new = floor(df.tune_new / (TUNE_MULT*1000)) * (TUNE_MULT*1000);	// set last three digits to "0"
	UiDriver_FrequencyUpdateLOandDisplay(true);
}
```
As you can see, no difference is made between OVI40 and mcHF. Same code.
And TUNE_MULT is on all machines 4. 

@df8oe : The Si5351a difference x1/x4 is hidden inside the Si5351a LO code.

@DL8EBD:  Most likely cause for confusion: You have different CW offset modes configured for mcHF and OVI40

DL8EBD, 2018-01-30:
> Most likely cause for confusion: You have different CW offset modes configured for mcHF and OVI40

no Danilo, both are set to 650Hz default

s53dz, 2018-01-30:
Hi,

Probably you should check this setting:
`CW Freq. Offset ( MENU_CW_OFFSET_MODE)`

> TX: display is TX frequency if received frequency was zero-beated. DISP: display is RX frequency if received signal is matched to sidetone. SHIFT: LO shifts, display is RX frequency if signal is matched to sidetone.

73 Bojan

db4ple, 2018-01-30:
`CW Freq Offset`  is a mode setting not a frequency value (that is `CW Side/Offset Freq`:
https://github.com/df8oe/UHSDR/blob/active-devel/mchf-eclipse/support/ui/menu/ui_menu_structure_mdtable.md#cw-mode-settings-menu_cw

We should change the names of these menu entries. It is confusing, agreed.

DL8EBD, 2018-01-30:
starting with the new daily D2.7.76 no more error in rounding to zero....
thx for fixing
Thomas, DL8EBD

closed 2018-01-30 by DL8EBD