df8oe/UHSDR #1382: on v D2.7.74

G4IOG, 2018-01-30:
Since I moved to this version, on v7 mchf I have big freq readout characters, and when I look at dual scope/waterfall, the scope is so small it is unuseable. Any one else have this same problem?
The scope mode or waterfall mode by themselves work fine. This has only occured on dual mode since large frequency digits

df8oe, 2018-01-30:
You must adjust percentage of waterfall / scope. And yes: big frequency digits are a new feeature.

G4IOG, 2018-01-30:
Go on then Andreas, give me a clue where to find the adjustment. The only percent I see is the one for the filter bandwidth background brightness. Thanks, I see a lot of new programming in the code now. Its working really well.

sp9bsl, 2018-01-30:
Hi,
the very first thing you must do is touchscreen calibration. Then simply press and hold the frequency ruler to adjust its location. After this, save your setting with MENU long press.

73
Slawek

G4IOG, 2018-01-30:
screen calibration is fine. I did it again just now. I did not know there was a drag and drop option on the bar. I found it now and it is back to how I like it. Many thanks for your help Slawek. Super work.
73, Bob

closed 2018-01-30 by G4IOG