df8oe/UHSDR #1383: **other place for "test reports"**

df8oe, 2018-01-30:
Because of issue list here will explode if test reports become usual here we decided that we no longer want to have theme here. There is another reason, too: if it is an operators error you will start a "discussion" here.

But test reports are highly welcome! Please use the following link:
https://www.amateurfunk-sulingen.de/forum/index.php?board=19
Here you can write test reports and you can get hints from other UHSDR users. You can discuss enhancement ideas and will get help. Before you are allowed to write there **please register**.

Many thanks and welcome in our group!

lb5va, 2018-02-04:
Tried NR with the .89 now and it works really fine now. But I have two questions/request.
1) Shouldn't level 0 be like as it is off and then increase turning up from 1...
2) Is it nessassary to go up to 99?  Many turns. I think two turns should be enough, 40 steps.

Stig

DD4WH, 2018-02-04:
Hi Stig,
thanks for your suggestion. Next version will contain that!
73 Frank

DF9TS, 2018-03-02:
Should the "other places for test reports" be mentione here: https://github.com/df8oe/UHSDR/blob/active-devel/CONTRIBUTING.md

?

Gerd.

closed 2018-06-09 by df8oe