df8oe/UHSDR #1385: the wider shift in rtty together with x16 and x32 zoom draws in the button (boxes) area

DG9BFC, 2018-01-30:
when i use x16 and x32 zoom and turn the shift for rtty up or down the wider shift draws in the boxes on the left side  (i use xlate +12kc) .... maybe programmers did only check minus 12 kc offset??
so it is a "small" graphical bug
the thing is not only that it draws the lines for mark and space inside the boxes ... when changing mode back to usb, lsb, whatever these boxes are not drawn again (so the marker lines stay there even after mode change)
after toggling trough modes, toggling trough digital modes and also switching on and off the dsp then all boxes are redrawn ... but the marker lines stay there BETWEEN the boxes (so the small rest of the lines never go away)

df8oe, 2018-01-31:
I am not sure this is a bug. OK - I always run at -12KHz. But setting to +12KHz does not give the described behaviour here - all is working as expected (OVI40, 3.5" LCD). We need more testers - best with samll LCDs.

DG9BFC, 2018-01-31:
i can upload a picture to forum so you can see
the narrow shift 85hz with x32 draws in the boxes and the wide shift 850hz in x16 draws in the boxes
and those lines stay there even after switching shift back to other setting
boxes area is not redrawn or only when boxes content is changed but then the lines stay between boxes

sp9bsl, 2018-01-31:
Hi,
I've seen the staying lines problem yesterday. This is refresh problem after change. Fixup will follow.

DG9BFC, 2018-01-31:
it should NOT draw in the boxes ... that is not solved with refresh of the boxes area (boxes content is redrawn when you change filter or dsp setting or digital mode ... but then the lines stay between these boxes 
and cause only the boxes are redrawn that will not solve it (or do you speak about refresh of the complete area??) ... they should not be in the area at all
(or we need the  graphical 3 kc shift depending on sideband for zoom x16 and x32 ... that also should solve it but it was reported that this would need more ram)

sp9bsl, 2018-01-31:
No, I've seen the problem when swithing between RT-L and RT-U and this is refresh problem. The pictures you showed on Andreas forum is simply driver unification problem after weekend works. Thank you for this report! Fixup will follow for both issues, stay tuned :)

df8oe, 2018-01-31:
fixed

closed 2018-01-31 by df8oe

DG9BFC, 2018-01-31:
that was again a fast support :-)
thank you