df8oe/UHSDR #1387: S-meter

UR7FM, 2018-01-31:
S-meter work not proper, even without antenna show S3-S4. Fix this problem please.

df8oe, 2018-01-31:
At my rig there is S3 with 50R connected and that is correct. All other S values I have tested all are correct.
You see the resistor noise :)

DD4WH, 2018-01-31:
Please inform yourself about the concept of an S- meter
https://github.com/df8oe/UHSDR/wiki/S-Meter-&-dBm-Hz-display
The noise signal is the internal noise of your receiver
You can also calibrate tje Smeter if you have an accurate reference signal.
Also a noise measurement should be given in dbm/ Hz, because otherwise it is dependent on your filter bandwidth. So an Smeter measurement of noise without stating filter bandwidth is pointless.
Happy experimenting!
Frank DD4WH

UR7FM, 2018-01-31:
Thanks a lot for answer. Just checked with 50R dummy  with 300Hz filter S0 and -129dBm. Now clear how it work:)) I compaired with my IC7300 before. 73!

df8oe, 2018-01-31:
This project brings us all to think about the physics which stands behind of all :)

So I close this issue.

closed 2018-01-31 by df8oe