df8oe/UHSDR #1395: Incomprehensible power "off"

ly2bok, 2018-02-01:
Good afternoon, dear Programmers ,
I have a suspicion that the program has an unpleasant glitch. I do not think it's connected with the transceiver. Transceiver  SDR mcHF ver. 04
During the robot in the FT8 digit, the WSJT-X program with control of the CAT with the computer via USB (System XP or Win7) sometimes, it is strange, when switching to gear, the transceiver turns off. It will be necessary to check with what version it started. Before this, 100% were not. I work a lot with the FT8 digit on SDR mcHF. Tried on three computers, the result is the same. During one hour the transceiver turns off 2-3 times. Very unpleasant. After that, turn on the transceiver and everything again works until the next shutdown. Why this is still not clear. I think it's somehow related to CAT management. Who else noticed this. There is no overheating of the chips, I checked it. I will gather more information and write to you.
Justas LY2BOK

df8oe, 2018-02-01:
I cannot confirm here - and nobody has changed anything in CAT commands... We are waiting for more bug reports like this!

ly2bok, 2018-02-01:
The question is removed. The problem was in the Chinese USB cable.  Replaced and everything is OK !
Justas LY2BOK

closed 2018-02-01 by ly2bok