df8oe/UHSDR #1396: Where do I find the code for the MCU 512K

WM0I, 2018-02-01:
When I go to the CODE section so that I can see what has changed so that I can do test on the changes. All I find is the larger code for the MCU 1 M. Where do I find the code for MCU 512K?

Best regards
WM0I

DL8EBD, 2018-02-01:
https://www.amateurfunk-sulingen.de/uhsdr/firmware-latest/mchf/MCU_512KB/

73
Thomas

db4ple, 2018-02-01:
Hi,
I don't understand your question. There is no different source code for the 512k vs. larger machines. That would be not very wise.

What we  do: we turn off FreeDV by using a compile time switch. 

See this commit : https://github.com/df8oe/UHSDR/commit/a28c23fad0aaa5e6371a7987139ab1c8a7c6a0e3

By no defining USE_FREEDV we remove all code related to FreeDV from being used in the "small" build. This in turn produces a smaller binary. The compiler (and linker) simply do the job for us with having to change anything in the source.

WM0I, 2018-02-01:
The code on the daily commit has to different sizes, THe larger is 463KB the one for the MCU 512K is only 273KB. Go to the top of this page and click on <code> and that takes you to the larger code size.

db4ple, 2018-02-01:
Yes. I know. 
 
But what is the point of your question?

rensd, 2018-02-02:
Hi,

the code size also depends on the build system used compiler etc..
I build the mcHF code regulary on a Mac using the following

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors)
4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

Current github version 2.7.84 builds the full mcHF code to 409.945 bytes without
any modifications, and Bootloader to 31847 bytes.

So there is still away to run the full code on de 512K CPU. If the code
grow keeps the same pace is in the last week, nearly 15kB we should
run out in 2 weeks time.

Regards

Rens | PA3AXA

df8oe, 2018-02-02:
4.93 gcc is not an option for us because it is unable to build F7 or H7 binaries. We must use more recent toolchains :)

Of course you are free to use your old, personal gcc4.9 - but that is not "the official way".

closed 2018-02-02 by df8oe

rensd, 2018-02-02:
Clear,

do't know if any mcHF 512K F7/H7 exist but if as long as the code will fit in a mcHF 512K F4 i'm
happy to provide this full version just drop me an email.
 
Regards Rens | PA3AXA

df8oe, 2018-02-02:
No. McHF is stuck on F4 MCU. But UHSDR supports not only mcHF...

You can announce it in our discussion forum, too. Additionally you can attach a binary to your post so that others can access it.

vy 73
Andreas