df8oe/UHSDR #1401: Digital mode touch-problem.

lb5va, 2018-02-02:
Why does it not respond when I press the touch field for Digital mode.
I may press 3-10 time before it changes the mode. It beeps every time. Touch-panel IS calibrated.
Is this just the way it is?  (Tried 2.7.83 today.)

sp9bsl, 2018-02-02:
Hi,
definitely no. Make a simple test: powerup your rig with any key except touchscreen pressed - you will enter the key test screen. Int this mode you should be able to draw pixels by pressing the screen. You should see prcise line when you drag across the screen. Please check the area where the DIGITAL item is placed for accuracy. I suspect bad quality of your touchscreen.

73, Slawek

lb5va, 2018-02-04:
Thanks for this tips. I did and it had some glitches. I did new calibration, and it became much better.
Also line-drawing became more whole lines. 
Does not respond every time but almost now.  Thanks again.

lb5va, 2018-02-04:
Is it wise to do new touch calibration after every fw-upgrade?  It was some versions back I did it last. Therefor this problem I think.

sp9bsl, 2018-02-04:
Hi,
we did not change anything with this since end of December. You do not need to recalibrate the touchscreen unless you use back to deafult settings, because only this will desroy all touchscreen calibration data.

73, Slawek

closed 2018-02-05 by df8oe