df8oe/UHSDR #1406: Scrolling text message sometimes prints a character on  the next line

phaethon, 2018-02-03:
CW, PSK, RTTY decoding (i.e. UiDriver_TextMsgPutChar function) sometimes instead of scrolling a message prints a character on the next line. The character stays even if message cleared until overwritten by another character if it happens.

sp9bsl, 2018-02-03:
Hi Eriks,
yes, we know this annoying problem:) Will try to fix it in the near future ;-)

73, Slawek

db4ple, 2018-02-04:
I am very sure this is a race condition.

sp9bsl, 2018-02-04:
Yes, I think so, because there is nothing from "logical" check of this simple code.

sp9bsl, 2018-02-04:
I think, found it.

phaethon, 2018-02-04:
Please share the insights.

sp9bsl, 2018-02-04:
You will laugh... #1415

sp9bsl, 2018-02-04:
Works for 15 min without extra characters on CW.

sp9bsl, 2018-02-04:
Yes, you're right. It is defined separatelly but -1. So, we have the buffer overwrite for last"/0" somewhere.

sp9bsl, 2018-02-04:
Try something like [that](https://github.com/sp9bsl/UHSDR/commit/968e41569b521420446b4a4a8ca0d24c70ff4cd7). It also works for me...

df8oe, 2018-02-05:
What of both solutions should be merged?

sp9bsl, 2018-02-05:
Not sure for 100% it needs testing. I closed my yesterdays too early pull for this reason. Will sit at this today evening.

db4ple, 2018-02-05:
To be honest, I like none of these as these fight the symptoms not the cause. Some piece of code is overwriting the end of the data structure (which should read '\0') in some situations it seems.

sp9bsl, 2018-02-05:
yes - I think the same. Have to debug it more "deeply".

sp9bsl, 2018-02-06:
Hi,
I want to inform you that I still working on it. The good thing is, I can catch every occurence and debug it, the bad is that it is very rare and time consuming. Still analyzing the cause. One thing seems found: it is caused by "\n" somewhere...

db4ple, 2018-02-06:
Good finding, "\n". I'll have a look at it.

sp9bsl, 2018-02-09:
Is the last change resolved the problem? Please report.

phaethon, 2018-02-09:
After limited testing - issue seems to be gone.

closed 2018-02-09 by phaethon