df8oe/UHSDR #1416: Makefile build with GCC 7.2.1 (ARM GCC 2017 Q4) produces not working binaries, Eclipse works

db4ple, 2018-02-05:
We are facing an interesting problem: If we build with the newest GCC build provided by ARM (GCC 7.2.1 aka 2017-q4 release provided by ARM) under Linux Debian, we get a not working binary which runs into an unhandled interrupt during USB DEVICE initializatiion. Whereas a Eclipse build under Windows does not show these problems and works flawlessly.

Build options are are virtually identical. It is not yet clear which interrupt causes the problems since it is the "Default_Handler" which gets called and it is 20+ unhandled interrupts which go there. Need to dig deeper.
It is not the WWDG interrupt. I confirmed that by mapping this interrupt to a different routine. Need to do this for all other interrupts until we find the one. Not sure if this helps us. So far the only lead.
We should have someone build with Eclipse on Linux with the same GCC tool chain to see if it is a Linux problem oder Makefile problem (@df8oe ?)

rensd, 2018-02-05:
Hi,

made a build with gcc-arm-none-eabi-7-2017-q4-majoron OSX. It won't fit mine 512K mcHF
But if you want i can send you the binary its 455.096 bytes.

Regards Rens | PA3AXA

db4ple, 2018-02-05:
I think I fiound the issue and just commited a fix. Please read the explanation in the message of my pull request #1418. It seems that we hit a changed linker behavoir in gcc 7 vs. older gcc (at least if used in combination with -flto).

df8oe, 2018-02-05:
It is working now for all gccs. But Makefile builds with -O2 and -O3 differs in behaviour between older and newest compiler. -O2 works for gcc7 but not for older gccs - -O3 works for all. Very strange.

db4ple, 2018-02-05:
I will look into the -O2 / -O3 issue for gcc 6.3.1 and see if I can reproduce the problem.

db4ple, 2018-02-05:
Hi Andreas,
did you test a build with gcc 6.3.1 and the new Makefile? I was able to reproduce the problem with the 2.7.90 Makefile (ended in a hardfault) but after adding -g3 (making a debug build, it was gone). For that reason I could not really debug anything. 

Since the new Makefile does have -g3 already set, if not done, please test. Thank you.

db4ple, 2018-02-05:
@df8oe : I now tested a plain 2.7.91 build via makefile and gcc 6.3.1 and it works for me.

df8oe, 2018-02-05:
I have tested. Using -O2 firmware trapped in bootsplash (only gcc older than gcc7). Switching to -O3 it works (for all gccs).

db4ple, 2018-02-05:
Hm. Ok, I did not notice you changed -O2 back -O3.   Will now compile with -O2 and report.

db4ple, 2018-02-05:
Ok, I found the problem. It was an out-of-bounds error in the Hilbert Filter Init.
See #1421

df8oe, 2018-02-06:
Now everything is fine at all tested compilers. Good work!

closed 2018-02-06 by db4ple