df8oe/UHSDR #1417: DSP A-Notch

ly2bok, 2018-02-05:
V. 2.7.92 . SDR mcHF v.04 . Mem. 512.  DSP A-NOTCH  not working perfectly. Fully distorts the signal and makes a noise like a steam locomotive.
http://www.qrz.lt/ly2bok/Laikinas/A Noch.mp3
LY2BOK Justas

df8oe, 2018-02-05:
Cannot confirm. Works as expected on OVI40 and mcHF (STM32F429@2M].

Any other confirmations?

ly2bok, 2018-02-05:
How it sounds, so I wrote it down. The interference is suppressed well, but this distortion is also present on the NR, and on the A-NOTCH, NR + HOC
the sound  of the "locomotive" is added.

df8oe, 2018-02-05:
I cannot hear this here. Not at OVI40 and not at mcHF. What LCD and LCD mode do you have?

ly2bok, 2018-02-05:
HY28B   SPI  Controler  ILI9325

df8oe, 2018-02-05:
That may be the problem. SPI driven LCDs are much more noisy than parallel driven. I do own one mcHF using SPI LCD and there the noise is slightly to hear, too. The sound is known as "helicopter-noise". Best will be to drive LCD in parallel mode, I think... Not a firmware issue.

ly2bok, 2018-02-05:
Toutcscreen will work in the   LCD the parallel mode?

df8oe, 2018-02-05:
Yes, of course. LCD and touchscreen do have different SPI lines and can ba used independently.

ly2bok, 2018-02-05:
OK I'll do it.Thank.

mm0mhp, 2018-02-05:
I too have this low level helecopter noise evident on my mcHF, however on my radio it can be eliminated by setting NB to zero. (Its a clone radio, ver 0.4 board with 512K ) .

df8oe, 2018-02-05:
If you set NR to ZERO you switch it OFF. NR identifies this noise as "wanted signal" and craps it out of the noise floor.

Do you know if your LCD is in SPI or parallel mode? I am not sure but I think clones are working in SPI because they do have RTC... And they are "obsolete at buying". It is very bad to fit a 512KB MCU where it is known since early 2015 (!!!!) that 1M MCU is recommended...

mm0mhp, 2018-02-05:
I might not be describing this correctly - its the Noise Blanking (NB) that I sw to zero to stop the helecopter effect - I can still use noise reduction (NR) but now without the helecopter noise. I dont understand the relationship between the noise blanker and noise reduction and how they interact. The helecopter effect is only just above the general noise floor.

ly2bok, 2018-02-05:
A friend of the LCD has a parallel, the "helicopter" is audibly weaker, but the SSB signal is not very pleasant and slightly distorted. I did not long ago RTC, in parallel RTC does not work. Sorry.
    I'll have to go back to the old version

DD4WH, 2018-02-05:
Hi guys, thanks for your reports.
We have to sort this a little bit:
1.) noise reduction NR: the sound recorded by Justas is NOT the usual helicopter sound (digital noise from the display). It sounds completely different. Seems to me a sound indicating that the processor is overloaded and can only process parts of one FFT frame. (maybe I have a solution for that, but I need to know which filter and which demod_mode you used for your tests . . .)

2.) noise blanker: the description sounds like the real "helicopter" noise. I will try to sort that out, maybe the noise blanker is not capable of using the decimated sample rate of 6ksps.

We need the following information:
@JUSTAS: hardware type? demod_mode? filter_bandwidth? noise blanker must be OFF. 
@mm0mhp: hardware type? demod_mode? filter_bandwidth? please do the following tests:
- noise blanker OFF --> set to 0.
- NR on: how does it sound?
- NR off: autonotch on: how does it sound?
- NR on AND autonotch on: how does it sound?

Please for both of you: try to do the tests above in two conditions: with filter 2k7 BPF and the same again with 2k9 BPF. Use either LSB or USB.

Thanks in advance!

All the best 73s 
Frank

P.S.: please also note that I personally will not accept (and react) on issue reports based on "cloned" hardware

df8oe, 2018-02-05:
I have checked again. No such noise under all circumstances that are part of my standard working conditions. Sorry: I cannot contribute anything here...

mm0mhp, 2018-02-05:
Firmware 2.7.91
Hardware Ver 4 board. with RTC
Display HY28B SPI
Controller ILI9325
T/Screen XPT 2046
512K flash

Demod type LSB (on the 80m band).

2.7K BPF
noise blanker OFF --> set to 0.
NR on: how does it sound? .......V Good
NR off: autonotch on: how does it sound?..... V Good
NR on AND autonotch on: how does it sound? .... V Good

2.9K BPF
noise blanker OFF --> set to 0.
NR on: how does it sound? ....... poor (with low level helicopter)
NR off: autonotch on: how does it sound? ..... V Good as above in 2.7K
NR on AND autonotch on: how does it sound?...... V Poor (with helicopter)

73
MM0MHP

DD4WH, 2018-02-05:
Hi Harry,
thanks a lot for your detailed report!
So, what does this tell us . . .
* basically, the NR works on the small mcHF even with autonotch simultaneously enabled (@2k7 filter bandwidth -> which means at internally 6ksps)
* I do not understand, why the sound is poor with your hardware above 2k7 filter bandwidth . . . could be that:
1.) your hardware has such a heavy noise because of SPI (as Andreas already said) or because it is a clone or that the processor load (which is twice as high for the NR above 2k7 filter bandwidth) is not sufficient . . .
 Would be good to hear whether other mcHFs also have this phenomenon.

df8oe, 2018-02-05:
I am nearly sure it is SPI LCD noise...
RTC *and* low-level-noise is impossible on mcHF because not enough free GPIOs to get both.

Need more confirmations!

mm0mhp, 2018-02-05:
Ok thank you, just one more point which may help...
2.7K BPF LSB
NB off
DSP A Notch - V Good
NB ON
DSP A Notch - has helicopter effect

as you say needs more reports.... but on my radio the Noise Blanker setting seems key to the helicopter effect.

sp9bsl, 2018-02-05:
Hi,
we can check if this comes from lcd interface: when you hear the "helicopter effect" please press the MENU button to stop refresh the lcd. Plese report if this stops or changes this sound.

mm0mhp, 2018-02-05:
Here's a youtube clip switching the Noise blanker in and out.

https://www.youtube.com/edit?o=U&video_id=EwgQN-68pSk

mm0mhp, 2018-02-05:
Selecting menu during helicopter effect
https://www.youtube.com/watch?v=MBeYnVg87ZA

df8oe, 2018-02-06:
Very good demonstration that helicopter noise has its origin in hardware: noise of SPI LCD signals! Please all others who do have this effects test with MENU, too. If noise disappears, reason is hardware issue. Best solution:
- remove RTC mod (if applied)
- plug off LCD (if it is on headers as recommended)
- set jumpers on LCD to "parallel drive"
- remove R30...R32 on ui pcb for removing SPI driving from HY28B headers
- switch LCD to parallel mode
- enjoy :)

If you can confirm this plaese close this issue.

ly2bok, 2018-02-06:
Еnjoy :) Without RTC :)  
Why install RTC, and then delete them. RTC in LCD very fine.This is not quite the right decision. In this version 2.5.130. "A-NOTCH" works perfectly, there is no noise and distortion.
https://youtu.be/2xgaEWhSJys
Justas LY2BOK

DD4WH, 2018-02-06:
Hi Justas,

yes, if it works perfectly for you, then use version 2.5.130! Thats why we have those old versions on the web.

To enjoy the newly implemented goodies, you would have to change your hardware (and subsequently the firmware version).

73 Frank

ly2bok, 2018-02-06:
Hmm ..... I think differently.
Ok thanks

df8oe, 2018-02-06:
Hardware design of mcHF mainly has not changed for some years (except minor changes). But firmware does step forward exploring new functionality. So there are three possibilities:

1) stuck on given hardware and possibly miss new features which do not work properly (or not) on old hardware
2) follow hints to get "best off" with old hardware (e.g. swap 512KB MCU, change LCD drive to parallel)
3) take a hardware which can do all at the same time

It is impossible to mix these three up.

DD4WH, 2018-02-06:
The "problem" with the noise blanker has also been identified.
It was related to the fact that a noise blanker setting of 1 was the strongest setting (higher numbers related to weaker noise blanker!) and is probably unsuitable for most receive situations.
We now reversed the numbers for the noise blanker setting:

0 = no NB activated
1 = weakest setting
9 = ideal setting for most situations with impulse noise
15 = strongest setting --> will probably lead to massive distortion in most cases

Attention! With no obvious impulse noise present in the audio, always set the NB to 0 = zero = OFF

Will be available in the next daily build.

Thanks Justas & Harry & Slawek & Andreas for your tests and reports!

73 Frank

mm0mhp, 2018-02-06:
Appriciate - thank you.
I plan to build a new UI board rather than make changes to the existing board, OV140 seems the natural choice.
73
Harry

df8oe, 2018-02-06:
More investigations done....

Please test using magnify=1. Does the helicopter noise remain?

df8oe, 2018-02-06:
Seems there are TWO different reasons for similar sound issues...

tinez99, 2018-02-06:
When there is distortion from the noise blanker, the work of the valcoder is also difficult. He's slowing down.

tinez99, 2018-02-06:
When there is distortion from the noise blanker, the work of the frequency encoder is also difficult. He's slowing down. encoder

tinez99, 2018-02-06:
When there is distortion from the noise blanker, the work of the frequency encoder is also difficult. He's slowing down.

mm0mhp, 2018-02-06:
loaded Ver 2.7.94
Yes helicopter noise evident with DSP auto notch and NB set to 1.
https://www.youtube.com/watch?v=vghoKaCjG4U

73
H

mm0mhp, 2018-02-06:
and noise with magnify settings = x1 and beyond.
also some evidence of a chopping noise when menu is selected.
https://www.youtube.com/watch?v=859gx_HUhy0

H

df8oe, 2018-02-07:
There are two different causes for the noise:

1) Hardware issue. The noise is more or less present in all working conditions. It stays when NB/NR/Notch is OFF and is present on all magnify modes. It differs between bands and/or frequencies. Mostly to hear with activated scope or waterfall, in some very bad combinations of UI and RF it can be heard in menu, too. It differs due to used components, cases and kind of LCD is driven (parallel or SPI). This cannot be removed by firmware.

2) Small MCU (F4) reaches its limits when
- magnify > 1 **and**
- NB is activated **and**
- autonotch is activated

We are not sure if we can remove this problem without cutting off something. Maybe all together is impossible on F4. We can implement a "brake" that if someone activates NB **and** autonotch automatically magnify jumps to "1". But there may be scenarios where it would be wiser to deactivate NB instead of settting magnify to "1". So it is users decision what he wants.

mm0mhp, 2018-02-07:
Ok understood, thank you for the clear explanation.

73
H

DG9BFC, 2018-02-07:
hmmm i am testing 2.7.94 .. with nb set to 10, nr set to 65, autonotch active ... spectrum AND waterfall active ... zoomed to x8 ... and it sounds just wonderful .... so seems as cpu is not reaching its limits even if you engage all gimmicks

DG9BFC, 2018-02-07:
just tested some more ... seems as autonotch alone needs more cpu as autonotch with nr together?!?

with autonotch alone (plus nb active) the limit between crackling noise (dirsturbed audio) and no crackling (clearly sign of cpu overload) is between 1.6 filter and 1.8 filter (2.3 and above unusable)

with autonotch plus nr active (and nb ... same as above) the border is between 3.2 and 3.4 kc filter
(so the 2.7 filter works fine)

all tested with zoom x8

when zoom set to x1 i can use one or two steps wider filter (crackling then starts between 3.6 and 3.8 filter with nr+notch)

... should not the autonotch alone produce LOWER cpu load?!?

anyway ... it works fine with normal bw settings and not too high zoom ratios ... will play some more with the notch settings in the menue ... hopefully i can further lower the cpu load a bit ... then no crackling should be heard ... or just with widest bw settings that you will never use when you have noise

mm0mhp, 2018-02-07:
Ive tried your settings but Im not seeing the same results here, 
1/ with NR off, and display at x8, NR at 95, the max bandwith before onset of helicopter noise is 2.7Khz
2/ Auto Notch is clean all B/Widths until NB is sw on - its a different noise but same basic helicopter effect.
3/ NR + NOTC, NB = 0, again clean till filter BW reaches 2.9Khz
Heres' the youtube of above....
https://www.youtube.com/watch?v=57YXJ7K8nck

73
H

df8oe, 2018-02-07:
It seems to be a *very critical* section. Settings which are not mentioned may impact, too. We have mainly identified the reason. Actually we are not sure that we can fix it for F4 (on OVI40 using f/ this issue does not exist). But we are trying to fix it. May take some days and some daily snapshots - we will see. Now there are three programmers (except me - who is much busy with OVI40 RF hardware) working together...

vy 73
Andreas

mm0mhp, 2018-02-07:
Yes much appriciate the effort going into solving for the F4 MCU when the later MCUs are not seeing the problem, I do intend building another UI board to get away from the F4 and 512K limits.

73
Harry

DD4WH, 2018-02-07:
Hi Harry,
would you mind testing again ;-):
Please set these settings in CONFIGURATION MENU
DSP Notch ConvRate    10
DSP Notch BufLen        128
DSP Notch FIRNumTap 32

and then test the following
NR + NOTC, NB = 0, BW 2k7
NR + NOTC, NB = 0, BW 2k9 and higher
Thanks in advance!

df8oe, 2018-02-08:
And again reports are needed.... Please test version 2.7.95 and report.

mm0mhp, 2018-02-08:
Morning:
Much appriciate all the effort you are putting in on this..... 
Here's the results with 2.7.95
https://www.youtube.com/watch?v=pvziEqHnvow
NB = 0, NR +NOTC, 2k7, no helicopter noise
NB = 0, NR +NOTC, 2k9, Helicopter noise but seem slower beat now? beat gets faster as BW incressed (sorry didnt video that part)

73
Harry

df8oe, 2018-02-08:
I think your are resting on the hardware issue now. The noise you are hearing is "received SPI noise". That is the part we cannot do anything in firmware - it must be eliminated in hardware. I have tested intensively on mcHF and can confirm that the software issue is completely gone now. Thank you for your observations! They are confirming this.

DG9BFC, 2018-02-09:
i only hear such hiss noise when i have no antenna connected ... and switching agc from off to fast ... then i have that hiss (but it is away after a second or so)

DG9BFC, 2018-02-09:
we all know that the display driven in spi does take more cpu load ... and the "helicopter noise" i nthe video is a clear sign of cpu overload ... i am sure when harry converts his display back to parallel driven the helicopter noise is gone ... so you have to decide what you want ... a real time clock and higher cpu load (causing helicopter noise when all things are engaged) ... or a lower noise level, no helicopter noise (but no rtc) ... i prefer lower noise and no helicopter ... and have a normal dcf77 driven clock on my desk

closed 2018-02-10 by df8oe