df8oe/UHSDR #1422: Meaning of System Menu text colors? (Question)

DF9TS, 2018-02-06:
What is the meaning behind system menu text colours for values?

Examples:
Menu -> Display Menu -> LCD Auto Blank  OFF (OFF in white text) 
Menu -> Display Menu -> Step Size Marker OFF (OFF in red text)
Menu->Configuration Menu -> Key Beep OFF (Off in orange text)

Would like to know the hidden logic behind this

BR

Gerd.

db4ple, 2018-02-06:
There is none. We "inherited" the coloring from the early authors Clint and Chris. It seemed to me that there was no consistent use of color schemes. When we cleaned up the menu code, we simply left the coloring as it was. 
I still hope someone gets so annoyed about this that he or she is going to change this and create a consistent use of colors in the menu.

DF9TS, 2018-02-06:
Tnx info

closed 2018-02-06 by DF9TS