df8oe/UHSDR #1425: S-meter starts getting red from S6 on, should be S9

DD4WH, 2018-02-06:
* My S-meter has red dots (instead of white dots) beginning with signal level S6
* from my point of view, the dots should all be white from S0 to S9 and then be red above S9

df8oe, 2018-02-07:
I do not have changing colours of dots at S6 or S9. Colour always stays grey (my setting). Observation on mcHF and OVI40. But there is a firmware detection of I/Q overload. If an overload is detected, complete (!) dots are coloured red for a short time and then return to standard colour. This mostly can be observed when strong spikes are received.

Or it is related to a setting or a combination of settings I do not have on any of my mcHFs / OVI40s...

DG9BFC, 2018-02-07:
on strong spikes the SCALE does change from white to read but NOT the s meter dots below the scale
they are white (my setting) also above S9

the +20 and +40 and maybe +60 dots could have a different colour for easier readout of higher signals but they are all one colour now

DD4WH, 2018-02-08:
digged a bit deeper:
* it seems that the red dots from S6 to S9 were an artefact produced during realtime adjustment of the colour of the dots in the menu (menu entry change does not redraw the whole S-meter :-))
* colour of the dots does not change from S9 to S9 + xdB --> is that intended?

Works now, so issue is closed.

closed 2018-02-08 by DD4WH